ros2/rclcpp #2334: change rclcpp logger sink or get all the sinks used by rclcpp

dxg397, 2023-10-11:
I have been trying to change the rclcpp logger sink to a rotating sink which is then be connected using fluentd to log to Elasticsearch. Is there any way to do this? I can do it if I used spdlog library but cannot do with external ROS libraries I am using.

- Operating System:
  -Ubuntu 22.04
- Installation type:
  - from source  using bazel
- Client library (if applicable):
  -  rclcpp::logger or MARCOS --> RCLCPP_INFO();

#### Expected behavior
the sink is defined as this:-
---> std::make_shared<spdlog::sinks::rotating_file_sink_mt>("/tmp/log/temp.log", 1048576 * 5, 1);

#### Actual behavior
spdlog is not able to find the loggers registered with rclcpp library even though it uses spdlog as the base.

## Feature request

Maybe having a conversion function to convert from rclcpp logger to base spdlog logger
spdlog::logger convertTospdLogger(rclcpp::logger)
{}

fujitatomoya, 2023-10-12:
> spdlog is not able to find the loggers registered with rclcpp library even though it uses spdlog as the base.

no, this is what we cannot do right now with [rcl_logging_spdlog](https://github.com/ros2/rcl_logging/tree/rolling/rcl_logging_spdlog). it statically uses `spdlog::sinks::basic_file_sink_mt` only.

https://github.com/ros2/rcl_logging/blob/014239dd80ab3de2a847af80f23d24da6438ecb7/rcl_logging_spdlog/src/rcl_logging_spdlog.cpp#L177

actually `rclcpp` (precisely `rcl`) does not know logging implementation but [rcl_logging_interface](https://github.com/ros2/rcl_logging/tree/rolling/rcl_logging_interface).
that means user can create own implementation to support those interfaces for ROS 2 logger. (we actually do this with our proprietary lock-free in-memory logger implementation.)

probably easiest way for you would be to replace `spdlog::sinks::basic_file_sink_mt` into `spdlog::sinks::rotating_file_sink_mt` in source code, and build it with your colcon workspace.

for long term, there is parameter argument `--log-config-file` with ros2 command to load the external logger configuration to the logger implementation. so that in `rcl_logging_xxx` can set up the configuration based on this config file. but it is not supported yet for `rcl_logging_spdlog`.

https://github.com/ros2/rcl_logging/blob/014239dd80ab3de2a847af80f23d24da6438ecb7/rcl_logging_spdlog/src/rcl_logging_spdlog.cpp#L107-L111

it would be appreciated if you consider the contribution, but we would want to have discussion what setting configuration should be supported for `rcl_logging_spdlog`.

fujitatomoya, 2023-10-12:
@clalancette i think we can move this issue to https://github.com/ros2/rcl_logging. what do you think?

jrutgeer, 2023-11-01:
> we actually do this with our proprietary lock-free in-memory logger implementation

Interesting. Any chance on open-sourcing that one?

fujitatomoya, 2023-11-02:
unfortunately no, we do not have any plan for that.