ros2/rclcpp #2340: multithreaded executor blocked when there are 8 running nodes

sgf201, 2023-10-26:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - < rclcpp-->

#### Steps to reproduce issue

```
launch 8 nodes for pull h264 stream and decode
launch 4 perception_node in multithread executor
```

#### Expected behavior
multithread executor works well

#### Actual behavior
multithreaded executor blocked
[image]

#### Additional information

----

Zard-C, 2023-10-26:
From the log, I found `process has died, exit -1`, could you check the progress status ? I don't think it's blocked.

fujitatomoya, 2023-10-26:
it looks like it received segmentation fault. but we need more information, could you provide how to reproduce this issue? e.g) distribution, rmw implementation, package version.

Zard-C, 2023-10-26:
> > From the log, I found `process has died, exit code -11`, could you check the progress status ? I don't think it's blocked.
> > It is blocked for a long while ,then crashed

could you add this detail to Bug Report? 

as you described:

1. The Executor was spinning but no callback was called, which was not expected to.
2. The Executor spining for a while (how long ?) then the the progress was recieved signal 11 (which is SIGSEGV) .

Out of my head, I have no explanation, why it was `blocked`, could you provide more information about  how to reproduce this issue?

Besides, you can run the program with gdb, and check where it crashed, maybe it's an important clue.

Zard-C, 2023-10-27:
@sgf201  any update 😄 ?