ros2/rclcpp #2348: Allow `rclcpp::Time::max()` to support other clock types than `RCL_SYSTEM_TIME`.

jrutgeer, 2023-11-01:
As of yet, `rclcpp::Time::max()` calls the `Time()` constructor without specified value for `rcl_clock_type_t`:

https://github.com/ros2/rclcpp/blob/fff009a75100f2afd8ef1c3863620bf5ebe67708/rclcpp/src/rclcpp/time.cpp#L278-L282

Hence, `rclcpp::Time::max()` returns a `Time` value with clock type `RCL_SYSTEM_TIME`:

https://github.com/ros2/rclcpp/blob/fff009a75100f2afd8ef1c3863620bf5ebe67708/rclcpp/include/rclcpp/time.hpp#L46

</br>

This means that you can't compare time values of type `RCL_ROS_TIME` to `rclcpp::Time::max()`.

I propose to change the `max()` implementation as follows:

```c++
Time
Time::max(rcl_clock_type_t clock_type = RCL_SYSTEM_TIME)
{
  return Time(std::numeric_limits<int32_t>::max(), 999999999, clock_type);
}
```

This is backwards-compatible with the current implementation, yet allows to specify `RCL_ROS_TIME` if needed, e.g.:

```c++
if( reported_time_ < rclcpp::Time::max(RCL_ROS_TIME) {
  [...]
}
```

fujitatomoya, 2023-11-03:
@jrutgeer it would be appreciated if you review https://github.com/ros2/rclcpp/issues/2348

fujitatomoya, 2023-11-05:
@jrutgeer this has been addressed by https://github.com/ros2/rclcpp/pull/2352, i will go ahead to close this. (i gave up backporting this to iron and humble, since this requires ABI change. see https://github.com/ros2/rclcpp/pull/2355#discussion_r1382609495)

closed 2023-11-05 by fujitatomoya