ros2/rclcpp #2350: /rosout logging without a node?

tylerjw, 2023-11-01:
I'm trying to get `/rosout` working for MoveIt 2.  At the last TSC meeting, I remember it being said that there might be a way to get `/rosout` logging working without a node. I'm here asking about that and providing the context of what I've tried to get logging working in ROS 1 as it did in ROS 2 for us. Here are my goals.

- logs publish to `/rosout`
- namespaced loggers that can have different log levels configured
- no large API changes (like adding a Node or Logger parameter to every constructor or free function)
- working on humble, iron, and rolling with a source build of our main branch

### Just the root node logger

To get around changing the API everywhere to take a Node I started with the idea that the logger could be stored in a static variable in a function and accessed through the return value of that function. Something like this is where I started:
```C++
rclcpp::Logger& moveit::get_logger() {
  static auto logger = rclcpp::make_logger("moveit");
  return logger;
}
```

Then wherever (usually in a main) a Node was created someone would then set the global logger:
```C++
auto node = std::make_shared<rclcpp::Node>("move_group");
moveit::get_logger() = node->get_logger();
```

Then everyone who wants to use it can just use that logger in their macro calls:
```C++
RCLCPP_WARN(moveit::get_logger(), "something went wrong, please help");
```

This works, but is short of the goal of namespaces.

### Try 1: Child loggers (member variables?)

To support child loggers I started with the idea that classes could create child loggers for themselves. I'd put something like this in the initializer list of the class:
```
, logger_(moveit::get_logger().get_child("fanuc_kdl_singular"))
```

This leads to the first problem:
```
[test_kinematics_plugin-1] [WARN] [1698723864.357328121] [rcl.logging_rosout]: Publisher already registered for node name: 'test_kinematics_plugin.fanuc_kdl_singular'. If this is due to multiple nodes with the same name, then all logs for the logger named 'fanuc_kdl_singular' will go out over the existing publisher. As soon as any node with that name is destructed, it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

> Note that this is on rolling, I know that using `get_child` on humble does not make a logger that publishes to `/rosout`. That is a sacrifice I'm willing to make.

This means that I can only have one of each child logger name in my program. So then we come to my second solution.

### Try 2: Child loggers as more function statics (works with free functions?)

As a result of the above warning and the need to log from inside free functions, I came up with another pattern.
```C++
// in moveit_utils library
rclcpp::Logger& get_logger() {
  static auto logger = rclcpp::get_logger("moveit");
  return logger;
}
```

```C++
// anonymous namespace at top of single cpp file
// delay creating the child until the first time some code in this file tries to log
rclcpp::Logger get_logger()
{
  static auto logger = moveit::get_logger().get_child("fanuc_kdl_singular");
  return logger;
}
```

This delays the creation of the child logger until the first time `get_child_logger` is called. But it results in a new problem. My tests crash after main in the clean-up of this new logger. 

I created an issue here with an example and stack trace: https://robotics.stackexchange.com/questions/105110/exception-thrown-in-rcl-logging-rosout-remove-sublogger-on-destruction-with-a-ch/105114#105114

@jrutgeer kindly responded with an explanation about what is happening and a fix to make the node itself have a static lifetime. Is this what people who use `/rosout` logging and child loggers are doing normally?

### Am I crazy?

Am I off with what I'm trying to do here, and there is some better way I'm just not seeing?

jrutgeer, 2023-11-01:
What about this:

```c++
void some_unrelated_function()
{
  RCLCPP_INFO(rclcpp::get_logger("node_name"), "Dut Node log"); // --> goes to /rosout
  RCLCPP_INFO(rclcpp::get_logger("node_name").get_child("some_child"), "Child log"); // --> goes to /rosout
}

int main(int argc, char** argv)
{
  rclcpp::init(argc, argv);
  rclcpp::Node::SharedPtr node = rclcpp::Node::make_shared("node_name");
 
  some_unrelated_function();

}
```

As long as the node named "node_name" exists, its logger can be accessed anywhere through its name.

tylerjw, 2023-11-01:
@jrutgeer the problem with that approach is I have many components in the MoveIt libraries that are used in many different possible nodes, including ones that do not exist in the MoveIt repo itself.

jrutgeer, 2023-11-01:
@tylerjw 

What I mean is the following:

The mapping between logger names and `/rosout` publishers is stored in **one** hashmap **per process**.

This means that if there are two nodes, named A and B, and node B has following code:

```c++
RCLCPP_INFO(rclcpp::get_logger("A"), "Some log message");
```
Then this will also be published through the `/rosout` publisher of node A, even though you are in node B.

And you don't have to be inside a node. Any function (e.g. not at all related to a node)  can call 

```c++
RCLCPP_INFO(rclcpp::get_logger("A"), "Some log message");
```

And this will still output to `/rosout` using the publisher of node A.

I think the only thing that is necessary is that:
- rclcpp::init() is called,
- Some node with a known name is present in that process for as long as you are publishing.

tylerjw, 2023-11-01:
Similar to what you are suggesting, I considered creating a node with the name moveit_{random number} in each process. Then each namespace (child logger) would be based on this. The problem is then I loose the information of which process the log came from in the rosout topic.

To continue your example let me rename A and B to `moveit_34` and `move_group` for a more concrete example.

The `move_group` node creates a `RobotTrajectory` type, which uses my `get_logger` function to create a child logger named `moveit_34.robot_trajectory`.

Separately I have another process with another node `moveit_task_constructor` that also creates a `RobotTrajectory` which then calls my logger function and that creates `moveit_12` and `moveit_12.robot_trajectory` loggers. 

When reading the logs I now know I have two different processes logging from within robot_trajectory but I do not know which one is which. This also presents new challenges around creating tooling that will call `set_logger_level` for specific sub-loggers to configure them to be at different log levels. I don't know before running the process what the root log node name will be because it has a random number associated with it to avoid collisions. And after I've started the process I'm unsure how I would figure out from what process came what node names so I can associate the generic moveit node with the named one.

jrutgeer, 2023-11-01:
What if you'd:
- Give the root node in each process always the same default name (e.g. "moveit_rosout"),
- The first process uses:
  `get_logger("moveit_rosout").get_child("move_group").get_child("RobotTrajectory")`
- The second process uses:
   `get_logger("moveit_rosout").get_child("moveit_task_constructor").get_child("RobotTrajectory")`

tylerjw, 2023-11-01:
> EDIT: you will indeed not be able to set the loglevel through a service if they all have the same name.

I hadn't thought of that but I do think I need a unique root node name for the service to work. I also need to spin whatever node starts with the service for setting log levels. I'm not sure if you can set log levels for loggers under a different root name in a process.

tylerjw, 2023-11-01:
Here is my prototype of your pattern. It does publish to /rosout (most of the time, I don't think it consistently flushes publish events, If I add a sleep at the end it always works):

```C++
#include <string>
#include <memory>
#include <unordered_map>
#include <rclcpp/rclcpp.hpp>

rclcpp::Logger& get_root_logger() {
  static auto node = std::make_shared<rclcpp::Node>("moveit");
  static auto logger = node->get_logger();
  return logger;
}

void set_node_logger_name(std::string const& name) {
  get_root_logger() = get_root_logger().get_child(name);
}

rclcpp::Logger get_logger(std::string const& name) {
  return get_root_logger().get_child(name);
}

int main(int argc, char** argv)
{
  rclcpp::init(argc, argv);
  rclcpp::Node::SharedPtr node = rclcpp::Node::make_shared("dut_node");
  set_node_logger_name(node->get_name());

  RCLCPP_ERROR(get_logger("child1"), "A");
  RCLCPP_ERROR(get_logger("child2"), "A");
  RCLCPP_ERROR(get_logger("child3"), "A");
  RCLCPP_ERROR(get_logger("child4"), "A");

  RCLCPP_ERROR(get_logger("child1"), "B");
  RCLCPP_ERROR(get_logger("child2"), "B");
  RCLCPP_ERROR(get_logger("child3"), "B");
  RCLCPP_ERROR(get_logger("child4"), "B");
}
```

jrutgeer, 2023-11-01:
Wrt. the code: 

- What is the "dut_node" for? 
- Other than that: looks good to me.

</br>

Wrt: the loglevel service:

- Similar to the `/rosout` hashmap, there is one loglevel hashmap per process (linking names to loglevels). So if you know the name of one node per process (that provides the set_loglevel_service), you can set loglevels for all loggers in that process (by their name).

-  Do realize that [this](https://github.com/ros2/ros2/issues/1355#issuecomment-1459900980) is, as of yet, still  an open issue. I am not so fond of using the set_loglevel_service because of that. It can lead to sporadic crashes.

tylerjw, 2023-11-01:
> Do realize that https://github.com/ros2/ros2/issues/1355#issuecomment-1459900980 is, as of yet, still an open issue. I am not so fond of using the set_loglevel_service because of that. It can lead to sporadic crashes.

I did not realize this. I was intending to try to re-create the file-based configuration of log levels we had in ROS 1. Is there any other ways you know of setting the log level per namespace except through the service?

tylerjw, 2023-11-01:
> What is the "dut_node" for?
dut stands for `device under test`. I have a python integration launch test that runs this and observes /rosout using rclpy.

tylerjw, 2023-11-01:
To be clear, the reason the above prototype works is the node used for publishing has `static` lifetime.

tylerjw, 2023-11-01:
I noticed on more thing. Each process creating a `moveit` node is going to cause a bunch of warnings about name collisions. To fix that, I updated the prototype to do something slightly different. I am now creating the node in the `set_node_logger_name` function.

This change has the side-affect that if anyone calls my `get_logger` function without first calling `set_node_logger_name` they are going to be greeted with an exception as `get_child` throws when it is called on a logger not associated with a node. I'll work on something to deal with that.

jrutgeer, 2023-11-02:
> Each process creating a moveit node is going to cause a bunch of warnings about name collisions.

Are you referring to [this warning message](https://github.com/ros2/rcl/blob/1b79535fa07fcc77c873adf5261308dac7a821ce/rcl/src/rcl/logging_rosout.c#L261-L265)?

If so, I don't understand it: 

That code is in `rcl_logging_rosout_init_publisher_for_node()`, which afaik is called only from the `node_base` constructor.
Even in your original code, shared_ptr to the node is static so it should be initialized only once?

So why would you get that warning, and why does it make a difference if the node is instantiated in `get_root_logger()` vs `set_node_logger_name()`?

</br>

> Is there any other ways you know of setting the log level per namespace except through the service?
As command line arguments, e.g.:

```
ros2 run package node --ros-args --log-level DEBUG --log-level rcl:=INFO 
```

tylerjw, 2023-11-03:
Thank you for looking into that warning message. I'm also confused by it, and it seems to have gone away with my latest changes based on the pattern above. I was probably doing something wrong before and creating multiple nodes with the same name somehow.

These changes led me to find another fun issue: https://github.com/ros2/rcutils/pull/434