ros2/rclcpp #2368: Component Container process has died ROS2 galactic RCLCPP

NamanMM, 2023-11-13:
Hello everyone,

I have been getting the following error (In the callback function of the node) resulting in the crash. It is sporadic and slightly tricky to reproduce but here is the error message:

1699879254.0570445 [ERROR] [component_container-1]: process has died [pid 597648, exit code -11, cmd '/opt/ros/galactic/lib/rclcpp_components/component_container --ros-args -r __node:=crop_spec_container -r __ns:=/'].

I have gone through the log files and nothing pops out immediately; the node seemed to be running well till this crash happens.

Any pointers or suggestion will be appreciated.
Thank you in advance.

clalancette, 2023-11-13:
Galactic is now End-of-Life.  Can you try this out on Humble, Iron, or Rolling to see if you still have the same problem?

Apurv354, 2023-11-14:
Try adding [gdb](https://www.sourceware.org/gdb/) to debug.

NamanMM, 2023-11-14:
I did and I am attaching the screenshot (Sorry I am not able to find the text where I copied the error). 
[image]

Zard-C, 2023-11-15:
From gdb: in frame 4, it was caused by std::vector<T>::push_back(T const &),  you can enter this frame and check the msg. My understanding of this issue is: `std::vector` is not thread-safe in the standard library, which means that simultaneous access to the same std::vector instance from multiple threads without appropriate synchronization mechanisms can lead to data races and memory corruption.

Zard-C, 2023-11-20:
any update? 😄  @NamanMM

mjcarroll, 2023-11-20:
What you are seeing is very unlikely an issue with the component container, but rather an issue with your node implementation.

While the component container (and components in general) can get you higher performance via intraprocess pub/sub, it comes with the downside of making a fault in a single node take down the entire container.

What I would suggest is that you separate the nodes that are in your component container and run them in separate process to make the system more robust.  You can then being to debug/diagnose the faulty node.

Looking at the traceback, I would be inclined to agree with @Zard-C that this is you trying to append to a non-thread-safe structure from multiple threads.

I'm going to close this out.  If you can establish that there is a bug in the component container, feel free to re-open.

closed 2023-11-20 by mjcarroll

NamanMM, 2023-11-20:
@Zard-C @mjcarroll Thank you guys. I did find an issue in how the vector was being used across threads - I have fixed it. I will monitor it for some time; happy to keep this closed.