ros2/rclcpp #2379: cannot use QoS override parameters via `NodeParameter` callbacks

fujitatomoya, 2023-12-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/1f5bd8ed43beea199dabe48bc8023af3aba9806c
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

- register callback that creates the subscription with `QosOverridingOptions` via `add_post_set_parameters_callback`.
- update parameter successfully, the gets `ParameterModifiedInCallbackException` exception.

#### Expected behavior

No exception received.

#### Actual behavior

`ParameterModifiedInCallbackException`

#### Additional information

This behavior blocks https://github.com/ros2/rclcpp/pull/2378

fujitatomoya, 2023-12-01:
this exception,

https://github.com/ros2/rclcpp/blob/26f6efa840cbac047a93dc21d055cfe9cde7270f/rclcpp/include/rclcpp/node_interfaces/node_parameters.hpp#L58-L80

comes from,

https://github.com/fujitatomoya/ros2_test_prover/blob/ab78461fa2fa9ce9f4cf5fe712a0fc307670028e/prover_rclcpp/src/rclcpp_2379.cpp#L52C1-L52C1

because it will call `declare_qos_parameters`, and this leads to call the parameter operations in the `NodeParameter` callbacks.

i think user expects that they can create publisher or subscription when the parameter is updated, i guess this is supported behavior. i am not sure if we really need `ParameterMutationRecursionGuard`...

@ros2/team what do you think?

clalancette, 2023-12-11:
> i think user expects that they can create publisher or subscription when the parameter is updated, i guess this is supported behavior. i am not sure if we really need `ParameterMutationRecursionGuard`...

We definitely needed it when I added it several years ago.  There was a race that could occur and that I was hitting often in the code I was working on.  But I'd have to go back and look at the PR that led to it to remember the situation.  Maybe there is another way to fix the race, but we shouldn't remove it without understanding the consequences.

fujitatomoya, 2023-12-11:
[External configurability of QoS policies](https://design.ros2.org/articles/qos_configurability.html) introduced after https://github.com/ros2/rclcpp/pull/781, user must be extremely careful with calling recursive function like this, otherwise it is really easy to loop forever. i think  we should keep `ParameterMutationRecursionGuard` behavior, on the other hand i think https://github.com/ros2/rclcpp/pull/2378 can bring good improvement for `rclcpp` application. besides, creating subscription on parameter change can be common use case.

more flexible and soft-restriction to have the maximum recursive count? I am not sure if this is good idea. i would like to hear other thoughts.