ros2/rclcpp #2383: Clock::sleep_for is not time shifts resistant

AlexeyMerzlyakov, 2023-12-07:
The `clock->sleep_for()` is not resistant to time shifts. This is related especially for backwards time jumps: when the time was changed to the past, the routine will hang for a long time (unless initially calculated due-time will be reached).

This I believe, is incorrect behavior for `sleep_for()`, which should sleep for a given period of time independently of what was changed in the system (e.g. time was corrected by OS time server, or something else).

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - ROS2 Rolling built from sources
- Version or commit hash:
  - fff009a75100f2afd8ef1c3863620bf5ebe67708
- DDS implementation:
  - FastDDS, CycloneDDS
- Client library (if applicable):
  - rclcpp/rcl

#### Steps to reproduce issue
The simple repro-testcase is attached to current ticket: [test_pkg.zip](https://github.com/ros2/rclcpp/files/13597086/test_pkg.zip).
It creates the node, clock and attaches the given clock to the node.
Then, `clock->sleep_for(2s)` is being called in a cycle, while execution time of each sleep are being calculated during run-time:
```
    rclcpp::Clock::SharedPtr clock = std::make_shared<rclcpp::Clock>(RCL_ROS_TIME);
    // Attaching clock to TimeSource, that subscribing to /clock and operating through this node
    rclcpp::TimeSource time_source(shared_from_this());
    time_source.attachClock(clock);

    while (rclcpp::ok()) {
      std::chrono::time_point<std::chrono::steady_clock> t1, t2;
      rclcpp::Time t1_ros, t2_ros;
      t1 = std::chrono::steady_clock::now();
      t1_ros = this->get_clock()->now();

      clock->sleep_for(2s);

      t2 = std::chrono::steady_clock::now();
      t2_ros = this->get_clock()->now();
      const auto dt = std::chrono::duration<float>(t2 - t1).count();
      const auto dt_ros = (t2_ros - t1_ros).seconds();
      RCLCPP_INFO(get_logger(), "Sleep time Real: %f ROS: %f\n", dt, dt_ros);
    }
```
When the clock is being shifted, `sleep_for()` will produce incorrect timings (for the forward clock shifts), or sleep for an indefinite periods of time (for backwards clock shifts).

To run the testcase, create testing workspace `mkdir -p ~/test_ws/src`, unpack there `test_pkg.zip` and build it there.

For verification on **System-time**, make sure that:
```
rclcpp::Clock::SharedPtr clock = std::make_shared<rclcpp::Clock>(RCL_ROS_TIME);
   was corrected to ->
rclcpp::Clock::SharedPtr clock = std::make_shared<rclcpp::Clock>(RCL_SYSTEM_TIME);
```
and run test binary:
```
./build/test_pkg/test_node
```
In parallel console, change the system time forwards or backwards as follows:
```
sudo timedatectl set-ntp 0
(today is Dec 7th)
sudo date -s "Dec 23"
sudo date -s "Oct 23"
sudo timedatectl set-ntp 1
```

For verification on **ROS-time**, clock should be of ROS-type:
```
 rclcpp::Clock::SharedPtr clock = std::make_shared<rclcpp::Clock>(RCL_ROS_TIME);
```
Run supplied by testcase launch-file, which runs Gazebo and testcase (using active `use_sim_time`) as follows:
```
ros2 launch test_pkg test_node.launch.py
```
In the Gazebo window, click "Reset Time" button in the bottom-right corner and see, that the testcase execution will hang.

#### Expected behavior

`clock->sleep_for()` works independently of changing system or ROS time, sleeping for a required periods of time.

#### Actual behavior

`clock->sleep_for()` sleeps for shorter periods of time when time was shifted forward, or for indefinite periods of time when time was shifted backward.

The following table shows the verification results for vanilla `Clock::sleep_for()` implementation:

| Vanilla rclcpp | Jump backward | Jump forward |
| ------------- | ---------- | ------------ |
| RCL_SYSTEM_TIME | Test hangs | Sleep-time was not corrected, but test works |
| RCL_STEADY_TIME | No influence | No influence |
| RCL_ROS_TIME, non-activated (no Gazebo and sim_time) | Test hangs | Sleep-time was not corrected, but test works |
| RCL_ROS_TIME, activated | Hangs  | No data |

#### Additional information

The problem appears because current implementation of `Clock::sleep_for()` - is just a wrapper for `Clock::sleep_until()`. In its body, `sleep_for()` calculates the initial "until" time as `now() + rel_time` and then sleeps unless this time to be reached. When the time (be it ROS-time or system time) was jumped, the routine will still sleep until the initially calculated time will be reached.

The time shifts respective implementation of `Clock::sleep_for()` might be considered to operate correctly irrespective from time jumps.

#### Implementation considerations

Since the time could be 3 types: system, steady and ROS, we need to handle correctly each type of time. Additionally, ROS clock could be active (subscribing for `/clock` messages) or inactive.

- For the cases when clock has system, steady or inactive ROS time, `sleep_for()` implementation could consider the sleep for a necessary amount of time, similar how does `rclcpp::sleep_for()` do it.  For all of these cases, the speed of time flaw will be normal, and be the same as how steady time flows.
- For the case when clock has active ROS time, `sleep_for()` should check the clock the same way as `sleep_until()` does it, but also respecting backwards time shifts in a clock jump callbacks.

So, the following overall implementation scheme could be:
```
bool
Clock::sleep_for(Duration rel_time, Context::SharedPtr context)
{
  ...
  if (this_clock_type == RCL_STEADY_TIME || this_clock_type == RCL_SYSTEM_TIME)
  {
    // Sleeping using std::chrono::steady_clock for time calculation
  } else if (this_clock_type == RCL_ROS_TIME) {
    if (!ros_time_is_active()) {
      // RCL_ROS_TIME is not active. Time flaws as steady one.
      // Sleeping using std::chrono::steady_clock for time calculation
    } else {
      // RCL_ROS_TIME is active. Need to sleep in respect to time jumps in the past.
    }
  }
}
```

The following patch [clock_20231207_2.patch.txt](https://github.com/ros2/rclcpp/files/13597451/clock_20231207_2.patch.txt) contains initial implementation of the proposed solution which gives the following result for testcase verification from above:

| Patch applied | Jump backward | Jump forward |
| ------------- | ---------- | ------------ |
| RCL_SYSTEM_TIME | No influence | No influence |
| RCL_STEADY_TIME | No influence | No influence |
| RCL_ROS_TIME, non-activated (no Gazebo and sim_time) | No influence | No influence |
| RCL_ROS_TIME, activated | No influence | No data, but it is expected to work w/o sleep time corrections |

Now, the `sleep_for()` works independently from time shifts. One case, when jumping forward in a Gazebo time is expected to sleep shorter periods of time. However, this is principal design problem, that I believe, could not be resolved at all:
There is no difference between forward time shifts (due to user or time-sync server) and time jumps obtained from `/clock` topic. So we could ignore this case.

clalancette, 2023-12-07:
Thank you for the detailed analysis!

> The following patch [clock_20231207_2.patch.txt](https://github.com/ros2/rclcpp/files/13597451/clock_20231207_2.patch.txt) contains initial implementation of the proposed solution

Can you please open a pull request with that patch?  Then we can more easily review it.  Thank you.

AlexeyMerzlyakov, 2024-01-29:
Closing the issue: current behavior is as expected (correct). More discussions could be found at the https://github.com/ros2/rclcpp/pull/2386#discussion_r1434316678 thread

closed 2024-01-29 by AlexeyMerzlyakov