ros2/rclcpp #2389: Make a Static Single Threaded Executor Node Container

KenYN, 2023-12-14:
## Feature request

#### Feature description

We can currently have in a Launch file:
```
    - node_container:
        pkg: rclcpp_components
        exec: component_container
```
or
```
    - node_container:
        pkg: rclcpp_components
        exec: component_container_mt
```
or:
```
    - node_container:
        pkg: rclcpp_components
        exec: component_container_isolated
```

It would be nice to also have for a `StaticSingleThreadedExecutor`:
```
    - node_container:
        pkg: rclcpp_components
        exec: component_container_static
```

#### Implementation considerations
I'm not sure if the component lifecycle would support this, or if not enough pre-existing nodes satisfy the requirements of creating all publishers, etc on startup.

Zard-C, 2023-12-14:
Hi @KenYN , you can start an discussion on https://discourse.ros.org/ first, then we can discuss the details rather than post an issue directly 😸

KenYN, 2023-12-22:
Given the `EventsExecutor` class that effectively supersedes these other executors, there's probably more of a need to address an `component_container_events` or adopt a whole new naming convention, so I'll close this as wontfix (needntfix?).

closed 2023-12-22 by KenYN