ros2/rclcpp #2393: Isolating Micro-ROS topics

Muthukumar4796, 2023-12-16:
When I add the line 'export ROS_DOMAIN_ID=47' to the bashrc file and then load Micro-ROS, my custom topics are not visible. However, when I comment out the ROS domain ID line, restart the agent, and reload Micro-ROS, my custom topics become visible. I'm stuck and unsure how to properly isolate my Micro-ROS topics without interference from the ROS domain ID and Can anyone provide guidance on how to properly isolate Micro-ROS topics in this setup?

Note:For Isolating the ROS2 topics using ROS_DOMAIN_ID is working fine.

Muthukumar4796, 2023-12-16:
Thanks for your response, Tobi. I did submit my query to the Micro ROS repository, but I also wanted to reach out to this community for any additional insights or support. Your positive feedback is greatly appreciated. Thanks again!

clalancette, 2023-12-20:
I'm going to close this, since it seems to be a query about Micro-ROS.  If this ends up being a problem with ROS itself, feel free to reopen.

closed 2023-12-20 by clalancette