ros2/rclcpp #2396: building with backward ros

jlack1987, 2023-12-21:
## Feature request

#### Feature description
We(nauticus robotics) have gotten in the habit of using [backward_ros](https://github.com/pal-robotics/backward_ros) when building our packages because it gives us a nice console trace when one of our nodes crash. In order to reduce the number of domain participants to improve DDS performance we have started combining nodes in composable node containers, but the composable node containers aren't built with backward ros so we lose the traces should something segfault which is really detrimental to finding/fixing those crash cases. Just curious if there's any way we could compile rclcpp with backward ros or some perhaps incorporating what backward ros does directly as I understand if adding an external dependency on backward_ros isn't desired for core packages like rclcpp.

mjcarroll, 2024-01-05:
I think that this is a reasonable feature request.  It is very helpful to get tracebacks, especially when you have multiple nodes in flight in the same container.  I think that we should probably make it an optional flag that you can turn on during build time (and probably make it opt-in, rather than opt-out).

I know that we already vendor `backward.hpp` inside of `osrf_testing_tools_cpp`: https://github.com/osrf/osrf_testing_tools_cpp/tree/rolling/osrf_testing_tools_cpp/src/memory_tools/vendor/bombela/backward-cpp

Maybe the best approach is to create a single vendor package that everyone can depend on?

Sushant-Chavan, 2025-08-04:
Hi @mjcarroll,

Just wanted to know if this ticket is already in your backlog.

I believe the fix involves adding a dependency to [backward_ros](https://github.com/pal-robotics/backward_ros) in the package.xml, and then adding a `find_package(backward_ros REQUIRED)` to the CMakelists.txt of `rclcpp_components`. I am not sure if there would be any issues in doing this, with rclcpp being one of the top-level pkgs. But looking at the [dependencies of backward_ros](https://github.com/pal-robotics/backward_ros/blob/foxy-devel/package.xml), I feel there shouldn't be much of a concern?

If its just updating the package.xml and CMakelists.txt, I can also make a PR for this. :)