ros2/rclcpp #2404: composition LoadNode does not set parameters for nodes in other namespaces

Aposhian, 2024-01-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - 16.0.7-1jammy.20231205.162058
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Some projects (in this case Nav2) have nodes that create other nodes that are not necessarily in the same namespace. When these nodes are loaded via composition, parameters passed to the `LoadNode` service are not propagated to these "sibling" nodes. This is a change in behavior from when not using composition: if you set parameters for the process, and then create any number of nodes, they all get their parameters.

```
ros2 run tf2_ros static_transform_publisher --frame-id map --child-frame-id base_link

ros2 run rclcpp_components component_container

ros2 service call /ComponentManager/_container/load_node composition_interfaces/srv/LoadNode "{ package_name: "nav2_planner", plugin_name: "nav2_planner::PlannerServer", parameters: [ {name: "global_costmap.global_costmap.resolution", value: { type: 3, double_value: "1.23"}}]}"
ros2 lifecycle set /planner_server configure
ros2 lifecycle set /planner_server activate
ros2 param get /global_costmap/global_costmap resolution
```

#### Expected behavior
`ros2 param get /global_costmap/global_costmap resolution` outputs `1.23`

#### Actual behavior
`ros2 param get /global_costmap/global_costmap resolution` outputs `0.1` (default)

#### Additional information
Setting parameters in the container works for this use-case. This may be a "won't fix" if this type of node creation is not desirable, and maybe could be replaced with sub nodes.
https://github.com/ros-planning/navigation2/issues/4011

SteveMacenski, 2024-01-10:
This issue is that sometimes a component node is itself a manually composed set of nodes. The other nodes within the component are not getting their parameters forwarded even though they're in the file passed to the component to load with the node. There's clearly a filter happening somewhere in the parameter namespaces, I think that should be relaxed. I don't see any downsides to passing a nodes more than it asked for, but there are downsides in passing less.

sloretz, 2024-01-18:
If I understand correctly, when components are loaded the parameters/remap rules/ros args are passed into the node via NodeOptions. See here where the `NodeOptions` are created:
https://github.com/ros2/rclcpp/blob/7901bb9da4e1addf71a307230afa60bd8d72f4f6/rclcpp_components/src/component_manager.cpp#L238

Outside of NodeOptions, the other way to get arguments is via the CLI. These are considered "Global arguments" because they apply to all nodes in the process. There's an option to disable them.

https://github.com/ros2/rclcpp/blob/7901bb9da4e1addf71a307230afa60bd8d72f4f6/rclcpp/src/rclcpp/node.cpp#L145-L147

I think global arguments on composable nodes are by default disabled. When loading composable nodes one normally doesn't want the node to get the parameters of the component manager, or of any other node loaded into the component manager. There is an option to allow it though:

https://github.com/ros2/rclcpp/blob/7901bb9da4e1addf71a307230afa60bd8d72f4f6/rclcpp_components/src/component_manager.cpp#L177-L187

> This issue is that sometimes a component node is itself a manually composed set of nodes. The other nodes within the component are not getting their parameters forwarded even though they're in the file passed to the component to load with the node.

I would recommend instead making the manually composed node pass the arguments from its own `NodeOptions` into the `NodeOptions` of the nodes it manually composes. Here's the API to get the options off the node that's doing the composition.
https://github.com/ros2/rclcpp/blob/7901bb9da4e1addf71a307230afa60bd8d72f4f6/rclcpp/src/rclcpp/node.cpp#L675-L678

SteveMacenski, 2024-01-23:
I grokk what you're saying, but not sure that addresses our problem - but happy to be wrong. This is the specific example we had in mind:
- `PlannerServer` is a composable node which can be used in a component container
- This node contains another node, [Costmap2D](https://github.com/ros-planning/navigation2/blob/main/nav2_planner/src/planner_server.cpp#L65)
- This object [is itself its own node](https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/costmap_2d_ros.cpp#L83-L106) to separate it from the planner server's logic, since it itself operates largely independently and is "force composed" into that node for efficiency / safety reasons

As you can see, we don't do anything with parameters / param file fed into the components from the launch file beyond a little bit of remapping of the node's name/namespace

Aposhian, 2024-01-23:
@sloretz are you suggesting that in the scenario that @SteveMacenski outlined, we could call `get_node_options` on the `PlannerServer`, and then feed that `NodeOptions` object when constructing the `Costmap2D`?

Since we are using `launch_ros`, we would also need to verify that it is not also filtering out parameters outside the namespace.

SteveMacenski, 2024-01-23:
@Aposhian do you want to prototype to make sure it solves the issue? Happy to merge something like that in. I could do it, but I wouldn't feel confident that all the items you're looking for are resolved as you're the reporter.

wjwwood, 2024-02-29:
Closing, based on discussion in our weekly issue triage meeting. Feel free to ask for it to be reopen if anyone thinks it needs more discussion.

SteveMacenski, 2024-02-29:
I think we're good, thanks

SteveMacenski, 2024-04-03:
I'm attempting to address this in Nav2 (https://github.com/ros-planning/navigation2/issues/4011#issuecomment-2033306272) but running into the issue that it doesn't appear to work by adding in the parent node's arguments and parameter overrides. 

@Aposhian tested by manually adding in a parameter into the `LoadNode` service and still failed, so is there a reason to believe that the NodeOptions is culling by namespace or node name on loading so that if passing a NodeOption object among other internal nodes with their own node names or namespaces it wouldn't be able to obtain its subset of information?