ros2/rclcpp #2416: STL Allocators deallocate method gets called with the wrong size

kfabian, 2024-01-26:
## Bug report
This is linked to #2220. I created this bug, because I wanted to show that it causes actual issues. If you think this is an unneeded duplicate, I apologize.
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Rolling
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I encountered the issue when linking against jemalloc, but it is a general issue, just hidden by the implementation of `operator delete(void* ptr, std::size_t size)` in the standard lib.
```
docker run -it --rm ros:rolling
apt update && apt install -y autoconf ros-rolling-demo-nodes-cpp
git clone https://github.com/jemalloc/jemalloc.git -b 5.2.1
cd jemalloc/
./autogen.sh --enable-debug
make dist && make -j && make install
export LD_PRELOAD=`jemalloc-config --libdir`/libjemalloc.so.`jemalloc-config --revision`
export MALLOC_CONF=tcache:false
source /opt/ros/rolling/setup.bash
ros2 run demo_nodes_cpp talker
```

#### Expected behavior
The node runs.
#### Actual behavior
The node crashed directly after startup.
#### Additional information
When an `rcl_allocator_t` struct is created from a std::allocator_traits-formatted Allocator using `get_rcl_allocator`, [the deallocate function will always be called with an element size of one](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/allocator/allocator_common.hpp#L67). [This](https://en.cppreference.com/w/cpp/memory/allocator/deallocate) is undefined behavior. The reason that this still works in the default case is, that `deallocate`, will call `operator delete(void* ptr, std::size_t size)`, which in the libstdc++ [implementation](https://github.com/gcc-mirror/gcc/blob/9693459e030977d6e906ea7eb587ed09ee4fddbd/libstdc%2B%2B-v3/libsupc%2B%2B/del_ops.cc#L31C8-L31C8) just calls `operator delete(void* ptr)` ignoring the size argument.

I think there is a fundamental issue that a STL Allocator  cannot be mapped to a `rcl_allocator_t` struct, because the deallocate method of the STL Allocator is size-aware where as the `rcl_allocator_t` struct ones is not.

clalancette, 2024-01-26:
Can you please just make a comment on #2220, showing your problematic behavior?

closed 2024-01-26 by clalancette

kfabian, 2024-01-26:
Everything is basically already in #2220, except the step-by-step instructions how it could be triggered.