ros2/rclcpp #2422: Heap Failure on deallocation of publisher/subsciber

NicoBrug, 2024-02-06:
Hi, 
I'm getting a heap error when deleting a publisher or subscriber.
My development environment :
- rclcpp humble
- Windows 10
- Unreal Engine
- Visual studio 2022

The behaviour is as follows: as soon as I create a subscriber or a publisher, there is no problem, it works normally. I can send and rcv message.
But when they are destroyed, following a request, closing the application, or go out of a scope I get a Heap Failure (Critical Error c0000374). 

debug With WinDBG
[image]

I investigated, and tried to make a custom allocator because I thought that the problem might come from where my allocation was made. 
The behaviour remains the same.

So I compiled rclcpp in RelWithDebug mode to access the symbol. And I found that the error seems to come from the deallocation of the EventHandler. 

[image]

In the image below, you can see that the event_handlers allocator is an allocator of type st::allocator<....>.

[image]

For your information, I've tested this on a non-Unreal environment (colcon build) and I haven't managed to reproduce this error. 

I'd like to know a little more about the behavior of custom allocators.

- I Was thinking in my understanding of rclcpp that a custom allocator would do all the allocating, isn't that the case?
- Have you ever seen this kind of behaviour?
- Could the fact that the event_handler uses a std::allocator whereas I use a custom one not be a problem? 
- Aren't event_handlers supposed to use the custom allocator? 

Too many questions come to mind!

It's quite complicated to debug this kind of thing and if you have any ideas I'd love to hear them. 

Thanks !

mjcarroll, 2024-03-27:
> I Was thinking in my understanding of rclcpp that a custom allocator would do all the allocating, isn't that the case?

Generally, this should be the case, but I don't believe that we have any test that rigorously guarantees that every piece is using the custom allocators.

> Could the fact that the event_handler uses a std::allocator whereas I use a custom one not be a problem?

This is the very likely scenario. 

> Aren't event_handlers supposed to use the custom allocator?

It should, but once again, there is a chance that it is not.

The best approach would probably be to start looking at the event_handlers and check to see if they are using the custom allocators like they are supposed to.  Outside of that, a minimal reproducable example would help a whole lot in investigating this.  Getting a Windows + Unreal setup is quite an undertaking.

kfabian, 2024-04-11:
Could this be related to #2220?

> > Could the fact that the event_handler uses a std::allocator whereas I use a custom one not be a problem?
> 
> This is the very likely scenario.

I think this should not cause an issue as long as each memory location is deallocated using the same allocator that was used for allocating it.