ros2/rclcpp #2435: :farmer: `TestGuardCondition.construction_and_destruction` throwing SEH Exception in Windows

Crola1702, 2024-02-23:
## Bug report

**Required Info:**

- Operating System:
  - Windows
- Installation type:
  - Source
- Version or commit hash:
  - Rollingo
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Crola1702, 2024-04-11:
Flaky ratio in the last 30 days (Apr 11th):

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rep|2024-04-09|2024-03-13|29|11|37.93

Crola1702, 2024-06-18:
Flaky ratio in the last 30 days (Jun 18th):

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_win_rep|2024-06-16|2024-05-21|25|8|32.0

ahcorde, 2024-07-18:
The line that is failing is https://github.com/ros2/rclcpp/blob/rolling/rclcpp/test/rclcpp/test_guard_condition.cpp#L61-L62

```cpp
    auto mock = mocking_utils::inject_on_return(
      "lib:rclcpp", rcl_guard_condition_fini, RCL_RET_ERROR);
```

ahcorde, 2024-07-18:
Segfault is happening in this line inside the `inject_on_return`

```cpp
mock_ = mmk_mock(target_c_str(), mock_type);
```

mjcarroll, 2024-07-18:
Thanks for isolating the area where the segfault is coming from!

is `target_c_str()` or `mock_type` null at this point?

ahcorde, 2024-07-18:
`target_c_str()` has some value (`lib:rclcpp`) but I'm still trying to understand what's inside the `mock_type`

ahcorde, 2024-07-24:
more details:

Code is failing here 

```cpp
mmk_vfprintf_ = (void*) plt_get_real_fn(ctx, vfprintf);
```

In particular `plt_get_real_fn` is calling `CreateToolhelp32Snapshot()` which is returning an `INVALID_HANLDE_VALUE` and the code abort.

ahcorde, 2024-07-24:
The error in particular is `The program issued a command but the command lenght is incorrect`

ahcorde, 2024-07-24:
Related fix https://github.com/ros2/Mimick/pull/38

clalancette, 2024-07-26:
Given what we merged yesterday, I'm going to consider that this one is "fixed", and close this out.  If it turns out not to fix the issue, please feel free to reopen.

closed 2024-07-26 by clalancette