ros2/rclcpp #2437: events-executor segfault on fast-dds `RMWSubscriptionEvent::update_data_available()`

alsora, 2024-02-24:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - ROS 2 iron 0.10.0-3
- DDS implementation:
  - Fast-DDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

This issue happens as part of a larger application, so I don't have a minimal reproducible example.
I have 2 nodes in the same process.
Each node has its own events-executor, spinning in a dedicated thread.

Node A periodically publishes an occupancy grid using this publisher
```
    rclcpp::PublisherOptions pub_options;
    pub_options.use_intra_process_comm = rclcpp::IntraProcessSetting::Disable;
    m_map_pub = parent_node->create_publisher<nav_msgs::msg::OccupancyGrid>(
      map_topic_name,
      rclcpp::QoS(rclcpp::KeepLast(1)).transient_local().reliable(),
      pub_options);
```

Node B from time to time creates a subscription to that topic, waits for a message and then lets the subscription go out of scope.
```
nav_msgs::msg::OccupancyGrid::ConstSharedPtr
get_occupancy_grid(
  const std::string & topic,
  std::chrono::seconds timeout)
{
  nav_msgs::msg::OccupancyGrid::ConstSharedPtr map;
  rclcpp::SubscriptionOptions sub_options;
  sub_options.use_intra_process_comm = rclcpp::IntraProcessSetting::Disable;
  auto map_sub = node->create_subscription<nav_msgs::msg::OccupancyGrid>(
    topic,
    rclcpp::QoS(rclcpp::KeepLast(1)).transient_local().reliable(),
    [&map](nav_msgs::msg::OccupancyGrid::ConstSharedPtr msg) {
      map = msg;
    },
    sub_options);

  auto start_time = std::chrono::steady_clock::now();
  while ((std::chrono::steady_clock::now() - start_time < timeout) && rclcpp::ok()) {
    if (map) {break;}
    std::this_thread::sleep_for(std::chrono::milliseconds(50));
  }
  return map;
}
```

#### Actual behavior

I have a unit-test that uses this setup, where the `get_occupancy_grid` function is called 4-5 times within the test duration.
The test fails sporadically, 1/30 times maybe, with the following backtrace
```
Thread 27 "test_positioner" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xffffc3ffe100 (LWP 23807)]
0x0000fffff7c65900 in ?? () from /opt/ros/iron/lib/librclcpp.so
(gdb) bt
#0  0x0000fffff7c65900 in ?? () from /opt/ros/iron/lib/librclcpp.so
#1  0x0000fffff7c62468 in ?? () from /opt/ros/iron/lib/librclcpp.so
#2  0x0000fffff7c661e4 in void rclcpp::detail::cpp_callback_trampoline<std::function<void (unsigned long)>, void const*, unsigned long, void>(void const*, unsigned long) () from /opt/ros/iron/lib/librclcpp.so
#3  0x0000fffff1d62a74 in RMWSubscriptionEvent::update_data_available() ()
   from /opt/ros/iron/lib/librmw_fastrtps_shared_cpp.so
#4  0x0000fffff1757428 in eprosima::fastdds::dds::DataReaderImpl::InnerDataReaderListener::on_data_available(eprosima::fastrtps::rtps::RTPSReader*, eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::SequenceNumber_t const&, eprosima::fastrtps::rtps::SequenceNumber_t const&, bool&) ()
   from /opt/ros/iron/lib/libfastrtps.so.2.10
#5  0x0000fffff16da1b8 in eprosima::fastrtps::rtps::StatefulReader::NotifyChanges(eprosima::fastrtps::rtps::WriterProxy*) () from /opt/ros/iron/lib/libfastrtps.so.2.10
#6  0x0000fffff16dcd30 in eprosima::fastrtps::rtps::StatefulReader::change_received(eprosima::fastrtps::rtps::CacheChange_t*, eprosima::fastrtps::rtps::WriterProxy*, unsigned long) ()
   from /opt/ros/iron/lib/libfastrtps.so.2.10
#7  0x0000fffff16dd28c in eprosima::fastrtps::rtps::StatefulReader::processDataMsg(eprosima::fastrtps::rtps::CacheChange_t*) () from /opt/ros/iron/lib/libfastrtps.so.2.10
#8  0x0000fffff16b6f4c in eprosima::fastrtps::rtps::StatefulWriter::deliver_sample_to_intraprocesses(eprosima::fastrtps::rtps::CacheChange_t*) () from /opt/ros/iron/lib/libfastrtps.so.2.10
#9  0x0000fffff16b7e5c in eprosima::fastrtps::rtps::StatefulWriter::deliver_sample_nts(eprosima::fastrtps::rtps::CacheChange_t*, eprosima::fastrtps::rtps::RTPSMessageGroup&, eprosima::fastrtps::rtps::LocatorSelectorSender&, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&) () from /opt/ros/iron/lib/libfastrtps.so.2.10
#10 0x0000fffff188c204 in ?? () from /opt/ros/iron/lib/libfastrtps.so.2.10
#11 0x0000fffff16b08d4 in eprosima::fastrtps::rtps::StatefulWriter::unsent_change_added_to_history(eprosima::fastrtps::rtps::CacheChange_t*, std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&) () from /opt/ros/iron/lib/libfastrtps.so.2.10
#12 0x0000fffff1735d8c in eprosima::fastdds::dds::DataWriterImpl::perform_create_new_change(eprosima::fastrtps::rtps::ChangeKind_t, void*, eprosima::fastrtps::rtps::WriteParams&, eprosima::fastrtps::rtps::InstanceHandle_t const&) () from /opt/ros/iron/lib/libfastrtps.so.2.10
#13 0x0000fffff1736694 in eprosima::fastdds::dds::DataWriterImpl::create_new_change_with_params(eprosima::fastrtps::rtps::ChangeKind_t, void*, eprosima::fastrtps::rtps::WriteParams&) ()
   from /opt/ros/iron/lib/libfastrtps.so.2.10
#14 0x0000fffff1736760 in eprosima::fastdds::dds::DataWriterImpl::create_new_change(eprosima::fastrtps::rtps::ChangeKind_t, void*) () from /opt/ros/iron/lib/libfastrtps.so.2.10
#15 0x0000fffff17283ac in eprosima::fastdds::dds::DataWriterImpl::write(void*) ()
   from /opt/ros/iron/lib/libfastrtps.so.2.10
#16 0x0000fffff1d6c5d8 in rmw_fastrtps_shared_cpp::__rmw_publish(char const*, rmw_publisher_s const*, void const*, rmw_publisher_allocation_s*) () from /opt/ros/iron/lib/librmw_fastrtps_shared_cpp.so
#17 0x0000fffff7e0806c in ?? () from /opt/ros/iron/lib/librcl.so
#18 0x0000ffffe45e2260 in rclcpp::Publisher<nav_msgs::msg::OccupancyGrid_<std::allocator<void> >, std::allocator<void> >::do_inter_process_publish (msg=..., this=0xaaaaab63dfe0)
    at /usr/include/c++/11/bits/shared_ptr_base.h:1295
#19 rclcpp::Publisher<nav_msgs::msg::OccupancyGrid_<std::allocator<void> >, std::allocator<void> >::publish<nav_msgs::msg::OccupancyGrid_<std::allocator<void> > > (msg=..., this=0xaaaaab63dfe0)
    at /opt/ros/iron/include/rclcpp/rclcpp/publisher.hpp:273
```

Quchaosheng, 2026-10-21:
I reproduced the lifetime mechanism in the current source, though not yet with a deterministic minimal test.

`SubscriptionBase` registers an RMW on-new-message callback via `set_on_new_message_callback()`, but `~SubscriptionBase()` currently does not explicitly clear it. The callback's `std::function` storage is therefore destroyed while the middleware can still hold its address; Fast DDS can invoke the trampoline from `RMWSubscriptionEvent::update_data_available()`, matching the reported backtrace.

A likely minimal fix is to call `clear_on_new_message_callback()` before the subscription handle and callback storage are destroyed, with destructor-safe exception handling. I have not submitted that patch because the callback unregister path can throw and because a deterministic test needs to assert the null-callback unregister at destruction, not merely rely on a probabilistic segfault. The existing `mimick` test infrastructure should make that test possible.

Would a test that intercepts `rcl_subscription_set_on_new_message_callback`, verifies the callback is unregistered before the subscription handle is destroyed, and exercises an in-flight callback boundary be acceptable as the regression shape? I can follow up with that test and a minimal destructor fix.

Quchaosheng, 2026-10-21:
The next step I can take is the mimick-based regression: assert that the RMW on-new-message callback is cleared before the subscription handle is destroyed, then stress an in-flight callback at that boundary. I will not rely on a probabilistic SIGSEGV as the test signal.

Assisted-by: OpenAI Codex (GPT-5.4), used for source review and test planning.

Quchaosheng, 2026-10-21:
Correction to my earlier comment: `on_new_message_callback_` is declared before `subscription_handle_`, so the subscription is destroyed first and the callback storage is destroyed afterward. That means the missing destructor clear is not, by itself, a use-after-free. I should not have stated it as the confirmed mechanism.

The remaining question is whether Fast DDS can still invoke the RMW callback after `rcl_subscription_fini()` has begun. That needs to be shown with the reproducer; if it cannot, the issue belongs to the RMW/DDS teardown path rather than to `~SubscriptionBase()`.

Assisted-by: OpenAI Codex (GPT-5.4), used for source review.