ros2/rclcpp #2447: the subscriber's destructor does not wait until currently processing callbacks are exited

GoesM, 2024-03-11:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 22.04
- Installation type:
  -   binaries
- Version or commit hash:
  -  the latest
- DDS implementation:
  - defaulted
- Client library (if applicable):
  -  rclcpp

#### Steps to reproduce issue

the subscriber's destructor does not wait until currently processing callbacks are exited, which may cause troubles like UAF, null-pointer bugs. 

Details are shown in navigation2 stack: https://github.com/ros-planning/navigation2/issues/4166

#### Expected behavior
We believe that the destructor of subscriber should have mechanisms to ensure complete release of relevant resources and fully exit ongoing works.

#### Actual behavior
there's no such mechanism. 

#### Additional information

If there is no wait mechanism implemented in the subscriber's destructor to ensure that currently processing callbacks are exited beforehand, it could introduce the possibility of the freed pointer being accessed by ongoing tasks within the subscriber.

Consequently, this scenario may lead to instances of **Use-After-Free (UAF)** and **null-pointer** bugs. One illustrative example can be found within the navigation2 stack, as documented in the following issue: https://github.com/ros-planning/navigation2/issues/4166

alsora, 2024-03-11:
This is just an initial thought, I haven't looked at the nav2 ticket.

Destructors in c++ must not be called while an object is in use.
The destructor shouldn't wait for callbacks to end, but rather it should not be invoked until the subscription is executing callbacks.

ROS already has mechanisms to ensure this. 
Subscriptions are managed through shared pointers, executors have weak ownership and will lock the pointer (thus preventing the destructor to be called) while executing a callback.

GoesM, 2024-03-11:
So, when we actively call the 'reset()' function of the shared pointer ( [an example in nav2](https://github.com/ros-planning/navigation2/blob/211273f831108a767a42ccc8bfc5d5de50761cc1/nav2_amcl/src/amcl_node.cpp#L329) ), if this subscription is executing a task at the same time, the subscription pointer would not be released immediately, but will only be released after the task is completely completed. May I understand it correctly?

mjcarroll, 2024-03-11:
I think there may be something problematic in the architecture of the node that you are running under ASAN.  I posted some observations on the original upstream issue: https://github.com/ros-planning/navigation2/issues/4166#issuecomment-1988937428

GoesM, 2024-03-11:
greatly thanks for your help ~

mjcarroll, 2024-03-21:
@GoesM since this is an upstream issue, I'm going to close it out.  Feel free to reopen (or open a new issue) if you find something else related here.

closed 2024-03-21 by mjcarroll

GoesM, 2024-03-21:
Thanks a lot, for all of your help and patience~