ros2/rclcpp #2461: New do_intra_process_publish compiler warning after upgrading rolling to Ubuntu 24.04

bijoua29, 2024-03-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - binary
- Version or commit hash:
  - latest rolling sync on 2024-03-13 (update for Ubuntu 24.04)
- DDS implementation:
  - Connext
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Compile application that uses rclcpp::Publisher and calls publish(). This includes custom application packages as well as ros2_control packages compiled from source.

#### Expected behavior
No compiler warning from rclcpp::experimental::IntraProcessManager::do_intra_process_publish

#### Actual behavior
Compiler warning from rclcpp::experimental::IntraProcessManager::do_intra_process_publish
. See https://gist.github.com/bijoua29/67c97d1210afda907d3a082cc55a915e

#### Additional information

fujitatomoya, 2024-03-21:
can we have the same warning with ubuntu 22.04? rolling official platform is currently 22.04, https://docs.ros.org/en/rolling/Installation.html

bijoua29, 2024-03-21:
This warning was not present in Ubuntu 22.04. I am confused though. I thought rolling was on [Ubuntu 24.04 now](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2024-03-13/36626)

clalancette, 2024-03-21:
> This warning was not present in Ubuntu 22.04. I am confused though. I thought rolling was on [Ubuntu 24.04 now](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2024-03-13/36626)

Sigh, yes.  We transitioned over the binary packages to Ubuntu 24.04, but because there are some bugs in 24.04, CI is still on 22.04.  That said, we are working towards moving CI over to 24.04 (hopefully next week), so we should fix any bugs we find on 24.04.

bijoua29, 2024-03-22:
Thanks for clarifying. I'm guessing this issue will come up in other places as well.

homalozoa, 2024-04-07:
@bijoua29 @clalancette I also got this warning in my project. All warnings are treated as errors in my project so I have to fix it.  https://github.com/ros2/rclcpp/pull/2492 is the commit that fixes the warning. Please check it and  try it.

bijoua29, 2024-04-08:
> @bijoua29 @clalancette I also got the same warning in my project. All warnings are treated as errors in my project so I have to fix it. #2492 is the commit that fixes the warning. Please check it and try it.

@homalozoa Do you mean you want me to try it by compiling rclcpp from source with your fixes. I'm afraid I'm not set up for compiling rclcpp from source

homalozoa, 2024-04-09:
> @homalozoa Do you mean you want me to try it by compiling rclcpp from source with your fixes? I'm afraid I'm not set up for compiling rclcpp from source

@bijoua29 No, you just need to modify the file I changed in this commit. Just one line.

bijoua29, 2024-04-09:
@homalozoa Ah sure. I will give it a try

fujitatomoya, 2024-04-12:
closing in favor of https://github.com/ros2/rclcpp/pull/2492

if you still find the problem, feel free to reopen or create another issue.

closed 2024-04-12 by fujitatomoya