ros2/rclcpp #2474: Bug in Executor::get_next_ready_executable ?

jmachowinski, 2024-04-02:
Hey

https://github.com/ros2/rclcpp/blob/f9c4894f96ea083fc5acfcea4f1ea83850759e63/rclcpp/src/rclcpp/executor.cpp#L692

I am highly confused by this code block:
```c++
        auto callback_group = entity_iter->second.callback_group.lock();
        if (callback_group && !callback_group->can_be_taken_from()) {
          continue;
        }
``` 

Shouldn't it be ?
```c++
        auto callback_group = entity_iter->second.callback_group.lock();
        if (!callback_group || !callback_group->can_be_taken_from()) {
          continue;
        }
```

jmachowinski, 2024-04-02:
@wjwwood @mjcarroll this was changed in #2142

mjcarroll, 2024-04-02:
You are correct, and the ones that follow are also incorrect, I will patch.

Good catch!

jmachowinski, 2024-04-17:
fixed

closed 2024-04-17 by jmachowinski