ros2/rclcpp #2480: TimersManager doesn't follow ROS time

alsora, 2024-04-03:
## Feature request

#### Feature description

Currently, the `TimersManager` class waits for timers to be ready using the following code

```
timers_cv_.wait_for(lock, time_to_sleep.value(), [this]() {return timers_updated_;});
```

This will always sleep using the system clock, even if the "time to sleep" was computed from a timer using the ROS clock

We should take into account the clock type when deciding how to sleep, and use an API that is aware of ROS clock

fujitatomoya, 2024-04-03:
duplication of https://github.com/ros2/rclcpp/issues/2457. lets keep this alive as tracking issue, and i will close https://github.com/ros2/rclcpp/issues/2457.
CC: @alsora

alsora, 2024-04-03:
Ops, my bad.
I started looking into this problem and created a ticket without noticing we already had one.