ros2/rclcpp #2508: Regression : Executor::spin_some_impl is active waiting

jmachowinski, 2024-04-17:
This was introduced in #2142

https://github.com/ros2/rclcpp/blob/3d58f0fb703d86461c2ab2fb39cd2640dc26e219/rclcpp/src/rclcpp/executor.cpp#L383

This condition make the executor active spin until the time is over if used in spin_all.

I guess 
```c++
    if (!wait_result_.has_value() || !exhaustive) {
      // In the case of spin some, then we can exit
      // In the case of spin all, then we will allow ourselves to wait again.
      break;
    }
``` 
is the correct implementation.

MichaelOrlov, 2024-04-25:
@wjwwood Could you please handle this issue? 
We decided to assign this issue to you at the latest waffle meeting. However, I don't have permission to change the Assignee field.

wjwwood, 2024-05-01:
Small update, I've iterated on the original pr to fix this here: https://github.com/ros2/rclcpp/pull/2517

Waiting to get it reviewed before merging it.

fujitatomoya, 2024-05-01:
@wjwwood i will try to taka a look in this week. thanks!

wjwwood, 2024-05-02:
Thanks @fujitatomoya, as always more reviews (even post merge reviews) are welcome, but @mjcarroll reviewed it already too.

wjwwood, 2024-05-02:
Closed with https://github.com/ros2/rclcpp/pull/2517

closed 2024-05-02 by wjwwood