ros2/rclcpp #2526: Backport PR2063 to Humble for Windows

BrettRD, 2024-05-10:
## Bug report
Building ROS2 Humble nodes on Windows, 
I have a complex third-party lib to build onto the side of a node, and it needs some flags set on the msvc compiler as a workaround for microsoft's defective macro expansions among other things.  Notably, `/std:c++20` is enough to trigger this bug.

With the flags I have to use, the windows compiler is pedantic about some namespace minutia I don't understand: `'type' is not a member of 'std'`  , and the `/permissive` flag has no effect.

Applying the patch from https://github.com/ros2/rclcpp/pull/2063 manually (just the 70 instances in humble/include/rclcpp/rclcpp/logging.hpp) allows the code to compile as expected.

This bug is not present in rolling, but I need my node to target Humble.

**Required Info:**

- Operating System:
Windows 11 Enterprise Evaluation
- Installation type:
Binary installation
- Version or commit hash:
  - 
- DDS implementation:
not relevant (fastrtps by default)
- Client library (if applicable):
rclcpp (logging.hpp)

#### Steps to reproduce issue

create a class derived from Node, and create a service server following the tutorial
Build using the flags
```
if (MSVC)
  add_compile_options(
    /Zc:preprocessor  # compliant preprocessor
    /std:c++20        # add support for __VA_OPT__
    /permissive
  )
endif()

```

#### Expected behavior
The node compiles

#### Actual behavior
msvc throws an error C2039 `'type' is not a member of 'std'`

#### Additional information
This is fixed by https://github.com/ros2/rclcpp/pull/2063

fujitatomoya, 2024-05-14:
https://github.com/ros2/rclcpp/pull/2529 is the backport PR to humble.

fujitatomoya, 2024-05-14:
closing in favor of https://github.com/ros2/rclcpp/pull/2529

closed 2024-05-14 by fujitatomoya