ros2/rclcpp #2555: ABI/API Compliance Checker in github workflow

fujitatomoya, 2024-06-08:
## Feature request

#### Feature description

Integrate ABI/API compliance checker in github workflow to check ABI compatibility for every PR to see if the fix can be backported automatically with label.

#### Background

Currently, for every PR, we (developers and maintainers) check the ABI compatibility if the PR is backportable to already available distribution. Sometimes it could be hard to tell with looking at the source code for sure, this also depends on the developers's skill and experience. (I personally run ABI compliance checker to make sure for this.)

Basically all bug fixes should be backported to the downstream distribution, but sometimes we miss and do that `need-to-do` or `requested` basis. (we can find comments requesting or asking for the backport to released distribution.)

To avoid this burden for developers and maintainers and keep the high quality for ROS distribution, it would be nice to have ABI compliance checker automatically run against the PR, so that we will be able to know the fix is ABI compatible with the tag or label provided by the workflow.

### Note

This `ABI Compatible` or `Backport-able` tag is an additional information for maintainers, says this does not mean we have to backport the fix to already released distros.
Although ABI is compatible, different behavior or significant different user-experience should be avoided breaking the user space.

#### Implementation considerations

- we can integrate https://github.com/lvc/abi-compliance-checker to github workflow. (or any other tools or recommendation?)
- build target branch and PR branch or repo to check the ABI compatibility for every commits.
- new label `ABI compatible` needs to be introduced and tagged by workflow.

I am creating this issue on rclcpp (because that is where i can find this situation most.) but the following core repositories can have the same ABI checker CI.

- https://github.com/ros2/rcl.git
- https://github.com/ros2/rcl_logging.git
- https://github.com/ros2/rcpputils.git
- https://github.com/ros2/rcutils.git
- https://github.com/ros2/rmw.git
- https://github.com/ros2/rmw_connextdds.git
- https://github.com/ros2/rmw_cyclonedds.git
- https://github.com/ros2/rmw_dds_common.git
- https://github.com/ros2/rmw_fastrtps.git
- https://github.com/ros2/rmw_implementation.git
- https://github.com/ros2/rosbag2.git

```mermaid
flowchart TD
    A[PR / Every Commit] -->|workflow| B[ABI Compliance Check]
    B --> C{ABI Compatible?}
    C -->|Yes| D[Tag / Label]
    D --> E[Code Review]
    C -->|No / Not Required| E[Code Review / Approval]
    C -->|ABI Compatible Required?| A[PR / Every Commit]
    E -->|Mergifyio| F[Backport Process]
```

after all, i think this will do some good for ROS users, developers and maintainers to accelerate the development process and ROS quality.

Tomoya,

clalancette, 2024-06-11:
I think that this is a good idea, though we'll have to be careful with it.  In particular, in `rolling` we allow ABI breaks at any time, while the released distributions don't allow ABI breaks.  We'll have to have configuration for that.

In addition, we have some ABI testing already available on the ROS buildfarm, though we haven't tested or enabled it since Galactic.  In particular, on your rosdistro repositories, you can add `test_abi: true`, which will do...something (see https://github.com/ros/rosdistro/blob/73d2581901d80429f1c9e371d2bf5de500d1d7dd/foxy/distribution.yaml#L4628).  So I think we should start by seeing what happens if we enable that (also, maybe @j-rivero , @nuclearsandwich , or @cottsay can shed some light on what that does, if anything).

fujitatomoya, 2026-08-16:
i have developed PoC actions,

- https://github.com/fujitatomoya/libabigail-action (base ABI checker action, application agnostic)
- https://github.com/fujitatomoya/ros2-abi-action (ROS 2 dedicated ABI/API checker)

see how it works,

- https://github.com/fujitatomoya/rcl_logging_syslog/pull/163 (ABI compatible label is tagged by action)

more verification is required.

fujitatomoya, 2026-10-14:
development log.

the abi compliance checker is now online for rolling, the followings are the related PRs.

- https://github.com/ros2/rclcpp/pull/3262
- https://github.com/ros2/rclcpp/pull/3276
- https://github.com/ros2/rclcpp/pull/3340
- https://github.com/ros2/rclcpp/pull/3341

> [!NOTE]
> if we want to backport the action to detect ABI compliance, we need to backport the above PR except tagging the backport labeling.

closed 2026-11-04 by fujitatomoya