ros2/rclcpp #2571: Action: rclcpp_action::Client goal response and feedback order not in send order

jmachowinski, 2024-07-01:
## Bug report

The order of the goal response and the feedback is not deterministic. 
In rare cases one gets the feedback callback before the goal response, that the goal was accepted.

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - source
- Version or commit hash:
  - Iron
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rcl / rclcpp

#### Steps to reproduce issue
Run a lot of actions, sometimes it happens.

#### Expected behavior
Order would be always the order as the server send the data.

#### Actual behavior
Order is mostly in send order, but sometimes not.

#### Additional information

There are multiple layers to this problem. 

AFAIK the goal itself and the response are send / received by a service call(Please correct me on this if this is a wrong assumption). The feedback is send through a second independent topic.
As we got two separate transports the order of the data is not guaranteed here.

The API in rcl does not expose the receive order:
https://github.com/ros2/rcl/blob/2b159af6dcecd0c9d17f59774e7e1b82cc7ef076/rcl_action/include/rcl_action/wait.h#L232
So even if the reception was deterministic, at this point the action client implementation just reorders, in case that multiple responses are available at the same time.

Barry-Xu-2018, 2024-07-02:
> AFAIK the goal itself and the response are send / received by a service call(Please correct me on this if this is a wrong assumption). The feedback is send through a second independent topic.
> As we got two separate transports the order of the data is not guaranteed here.

Right. Based on the current action design, the reception order (status and service response) on the user side cannot be guaranteed. Is this considered a bug ?

fujitatomoya, 2024-07-02:
similar issues https://github.com/ros2/rclcpp/issues/1679 and https://github.com/ros2/rclcpp/issues/2451

jmachowinski, 2024-07-02:
> Is this considered a bug ?

I would say yes, this is highly unexpected behavior, at least to me. We just ran into this, as we ignored feedback, before the goal was in the state accepted, which caused a bug in our application. Now we need to buffer it etc...

Barry-Xu-2018, 2024-07-02:
> I would say yes, this is highly unexpected behavior, at least to me. We just ran into this, as we ignored feedback, before the goal was in the state accepted, which caused a bug in our application. Now we need to buffer it etc...

Oh, indeed.  
I think it is the same issue described in https://github.com/ros2/rclcpp/issues/1679

jmachowinski, 2024-07-02:
I agree, seems to be the same bug as #1679, perhaps we should edit the title of it....
Closing as duplicate

closed 2024-07-02 by jmachowinski