ros2/rclcpp #2574: EventsExecutor doesn't support rcl guard conditions

alsora, 2024-07-02:
## Bug report

The `EventsExecutor` currently doesn't handle guard conditions managed in `rcl`.
An example is the graph guard condition.
The `EventsExecutor` won't receive events from these entities (note that `rclcpp::GuardCondition`s will still work).
Waitables that rely on them won't execute.

#### Additional information

We should either have rclcpp only deal with `rclcpp::GuardCondition` by wrapping them or propagate some callbacks from rclcpp down to the rmw to receive events.

mauropasse, 2024-07-04:
> We should either have rclcpp only deal with rclcpp::GuardCondition by wrapping them

This is the approach we have taken since the creation of `rclcpp::GuardCondition`. I was in charge of converting all `rclcpp` code using `rcl_guard_condition` to their `rclcpp::GuardCondition` wrapper.

alsora, 2024-07-06:
Yes, I agree that this is the best approach.
rclcpp users shouldn't have to deal with rcl objects.