ros2/rclcpp #2582: :farmer: Tests failing consistently in connext all distros: `test_client`, `test_server`, `test_subscription`

Crola1702, 2024-07-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Jammy and Noble
- Installation type:
  - Source
- Version or commit hash:
  - All distros
- DDS implementation:
  - RTI Connext only
- Client library (if applicable):
  - Rclcpp

#### Steps to reproduce issue

1. Run a CI build with RTI Connext enabled in any ROS2 distro
2. See test failures

#### Additional information

Reference builds: 
* https://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/80/
* https://build.ros2.org/view/Jci/job/Jci__nightly-connext_ubuntu_noble_amd64/31
* https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/
* https://build.ros2.orgi/job/Hci__nightly-connext_ubuntu_jammy_amd64/251/

Test regressions:
* [projectroot.test.rclcpp.test_client](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/projectroot.test/rclcpp/test_client/)
* [projectroot.test.rclcpp.test_service](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/projectroot.test/rclcpp/test_service/)
* [projectroot.test.rclcpp.test_subscription](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/projectroot.test/rclcpp/test_subscription/)
* [rclcpp.TestClient.on_new_response_callback](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/rclcpp/TestClient/on_new_response_callback/)
* [rclcpp.TestClient.client_qos](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/rclcpp/TestClient/client_qos/)
* [rclcpp.TestService.on_new_request_callback](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/rclcpp/TestService/on_new_request_callback/)
* [rclcpp.TestSubscription.on_new_message_callback](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/rclcpp/TestSubscription/on_new_message_callback/)
* [rclcpp.TestService.server_qos](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/rclcpp/TestService/server_qos/)
* [rclcpp.TestService.server_qos](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/rclcpp/TestService/server_qos/)
* [projectroot.test.rclcpp.test_executors](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/projectroot.test/rclcpp/test_executors/)
* [rclcpp.TestIntraprocessExecutors/EventsExecutor.testIntraprocessRetrigger](https://build.ros2.org/job/Ici__nightly-connext_ubuntu_jammy_amd64/144/testReport/junit/rclcpp/TestIntraprocessExecutors_EventsExecutor/testIntraprocessRetrigger/)
* [projectroot.test.rclcpp.test_client_common](https://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/80/testReport/junit/projectroot.test/rclcpp/test_client_common/) (in rolling)

Most of errors show an output similar to:
```
unknown file: Failure
C++ exception with description "failed to set the on new response callback for client: rmw_client_set_on_new_response_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:58" thrown in the test body.

unknown file: Failure
C++ exception with description "could not create client: error not set, at /tmp/ws/src/ros2/rcl/rcl/src/rcl/client.c:146" thrown in the test body.
```

<details>
<summary>
Full log output
</summary>

```
[ RUN      ] TestClient.on_new_response_callback
[ERROR] [1720517303.715179852] [rmw_connextdds]: rmw_client_set_on_new_response_callback not implemented
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
unknown file: Failure
C++ exception with description "failed to set the on new response callback for client: rmw_client_set_on_new_response_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:58" thrown in the test body.
[  FAILED  ] TestClient.on_new_response_callback (306 ms)

...

[ RUN      ] TestClient.client_qos
[ERROR] [1720517304.608630352] [rmw_connextdds]: failed to create DDS writer
[ERROR] [1720517304.608717614] [rmw_connextdds]: failed to create client requester
[... 32 lines cut ...]
C++ exception with description "could not create service: error not set, at /tmp/ws/src/ros2/rcl/rcl/src/rcl/service.c:159" thrown in the test body.
[  FAILED  ] TestService.server_qos (305 ms)

...

[ RUN      ] TestSubscription.on_new_message_callback
[ERROR] [1720517318.520322791] [rmw_connextdds]: rmw_subscription_set_on_new_message_callback not implemented
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
unknown file: Failure
C++ exception with description "failed to set the on new message callback for subscription: rmw_subscription_set_on_new_message_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:74" thrown in the test body.
[  FAILED  ] TestSubscription.on_new_message_callback (306 ms)

...

[ RUN      ] TestIntraprocessExecutors/EventsExecutor.testIntraprocessRetrigger
[ERROR] [1720517343.448656055] [rmw_connextdds]: rmw_subscription_set_on_new_message_callback not implemented
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
unknown file: Failure
C++ exception with description "failed to set the on new message callback for subscription: rmw_subscription_set_on_new_message_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:74" thrown in the test body.
[  FAILED  ] TestIntraprocessExecutors/EventsExecutor.testIntraprocessRetrigger, where TypeParam = rclcpp::experimental::executors::EventsExecutor (305 ms)
```

</details>

clalancette, 2024-07-11:
So I think this needs to be split into (at least) two separate issues.

In Iron and Humble, it is indeed the case that `rmw_client_set_on_new_response_callback` is not implemented, and that is the cause of the test failures that look like:

```
C++ exception with description "failed to set the on new response callback for client: rmw_client_set_on_new_response_callback not implemented, at /tmp/ws/src/ros2/rmw_connextdds/rmw_connextdds_common/src/common/rmw_listener.cpp:58" thrown in the test body.
```

I think that should be reported over at https://github.com/ros2/rmw_connextdds.

There is a separate issue here as well, which seems to happen on all distributions.  That problem looks like the following:

```
[ERROR] [1720178602.536775801] [rmw_connextdds]: failed to create DDS writer
[ERROR] [1720178602.536905075] [rmw_connextdds]: failed to create client requester
[ERROR] [1720178602.536920455] [rmw_connextdds]: failed to create RMW client implementation
[D0000|Pub(80000008)|T=rq/ns/empty_serviceRequest|CREATE Writer] DDS_LivelinessQosPolicy_is_consistentI:inconsistent QoS policy: assertions_per_lease_duration is set to more than 1000000 assertions per second
[D0000|Pub(80000008)|T=rq/ns/empty_serviceRequest|CREATE Writer] DDS_DataWriterQos_is_consistentI:inconsistent QoS policy: liveliness
[D0000|Pub(80000008)|T=rq/ns/empty_serviceRequest|CREATE Writer] DDS_Publisher_create_datawriter_disabledI:ERROR: Inconsistent QoS
unknown file: Failure
C++ exception with description "could not create client: error not set, at /tmp/ws/src/ros2/rcl/rcl/src/rcl/client.c:146" thrown in the test body.
```

That actually looks like a bug in the test, or at least the lease_duration looks bogus somehow.  That should be reported here in `rclcpp`; you could just rename this issue for that, or you could close this out and create a new one.

Crola1702, 2024-07-17:
This report was split between here and https://github.com/ros2/rmw_connextdds/issues/155

Crola1702, 2024-11-12:
Friendly ping @mjcarroll

clalancette, 2024-12-02:
I believe this was fixed by #2684.  I'm going to close this out, but feel free to reopen if this occurs again.

closed 2024-12-02 by clalancette