ros2/rclcpp #2606: ROS2 node having long name throws confusing error

prabeenkumar122, 2024-08-20:
I am trying to create a ros2 node using rclcpp apis. My node name is very long with all the namespaces. When I run, I am getting error 

sample node name:
node_ANHSSLFEHUHEGRCMYWOTSRRHOPUBGYUQRUHQBONVFXYIPNMJJCKGMREUXUMQLLBXCMKNYWBTMKUVYFVIKUJLDRSIJSYRDMQNEHYUXJGRJLWVKSHIIBXRGMTKGSTRPTNIBOFFENBMCLEOUDGGJKDKROJQRELXOOTGRIOQKGXLNQRLFRXJQWMQMDDDLXQXCIXBFTVPNXSRSVYYXHCUFBTNIOVQ

confusing error message:
Invalid service name: topic length should not exceed '255U - 8U':
      '/node_ANHSSLFEHUHEGRCMYWOTSRRHOPUBGYUQRUHQBONVFXYIPNMJJCKGMREUXUMQLLBXCMKNYWBTMKUVYFVIKUJLDRSIJSYRDMQNEHYUXJGRJLWVKSHIIBXRGMTKGSTRPTNIBOFFENBMCLEOUDGGJKDKROJQRELXOOTGRIOQKGXLNQRLFRXJQWMQMDDDLXQXCIXBFTVPNXSRSVYYXHCUFBTNIOVQ/set_parameters_atomically'

alsora, 2024-08-20:
Hi @prabeenkumar122 ,

This issue is not related to the `rclcpp` repo, but rather to `rmw.
That's where the warning comes from: https://github.com/ros2/rmw/blob/rolling/rmw/src/validate_full_topic_name.c#L148

However, why do you think it's so confusing? It indicates that the problem is that the names are too long.

sloretz, 2024-08-29:
Thank you for the report! We discussed this a bit today. In short, we're going to close the issue without making any changes.

While it's true that the node name is responsible for most of the topic name being too long, the validation function for the topic name is given the full name as a string. It can't tell that the node name is what's contributing to the excessive length.  The effort needed to make this error message clearer outweighs the benefit.