ros2/rclcpp #2607: Broken API document formatting for Publisher and ParameterEventHandler

EwingKang, 2024-08-21:
## Bug report

**Required Info:**

- Operating System:
  - **N/a** 
- Installation type:
  - **N/a**
- Version or commit hash:
  - rolling
- DDS implementation:
  - **N/a**
- Client library (if applicable):
  - rclcpp 

#### Steps to reproduce issue

Automatically generated documents from header files are broken in
* [parameter_event_handler.hpp#L95](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/parameter_event_handler.hpp#L95) with online API [link](https://docs.ros.org/en/iron/p/rclcpp/generated/classrclcpp_1_1ParameterEventHandler.html#_CPPv4N6rclcpp21ParameterEventHandlerE)
* [publisher.hpp#L190)](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/publisher.hpp#L190) with online API [link](http://docs.ros.org/en/rolling/p/rclcpp/generated/classrclcpp_1_1Publisher.html#_CPPv4N6rclcpp9Publisher21borrow_loaned_messageEv)

#### Expected behavior
* For the `parameter_event_handler`, all the example code in the comments becomes plain text in the generated html page, which is very hard to read. 
* For the `publisher`, the entire paragraph 
```publish the LoanedMessage instance is being returned to the middleware or free’d accordingly to the allocator. If the message is not being published but processed differently, the destructor of this class will either return the message to the middleware or deallocate it via the internal allocator.```
is wrap in the `see also` block, with weird breaking points, and the beginning of the phrase `With a call to` left outside of the block. This is probably not what we want

#### Actual behavior

#### Additional information
p.s I'm not sure if this counts as a bug or feature request

Screenshots - What I saw with my browser for `parameter_event_handler` and `publisher` respectively.
[image]

[image]

clalancette, 2024-08-21:
Thanks for the report.  See #2608, which should fix both of these issues.

closed 2024-08-21 by clalancette

EwingKang, 2024-08-22:
Thanks for the very very apt response!
I was trying to contribute to it, but couldn't find how to build the package specific document yesterday, despite that the [developer's guide stating that README should contain instruction to build documents](https://docs.ros.org/en/jazzy/The-ROS2-Project/Contributing/Developer-Guide.html#documentation). 
However, I do find it today under the [howto guides about Documenting a ROS2 package](https://docs.ros.org/en/jazzy/How-To-Guides/Documenting-a-ROS-2-Package.html) (haven't tried it, but I presume it is what I'm thinking)
Relavent:
* Here's the instruction for the [ros2doc](https://docs.ros.org/en/jazzy/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html)

p.s. The googlability seems to be really bad with https://docs.ros.org/en/, I have to navigate through it instead of landing in through searches.

PR #2608 (2024-08-21): Fix a couple of issues in the documentation.
These should make it render more nicely.

This fixes #2607