ros2/rclcpp #2610: A way to get the parsed global arguments from within the Node or NodeOptions

saikishor, 2024-08-22:
Hello!

i've a use case where I need to know the parsed global ros args inside the Node, but I see there is no current way as the NodeOptions arguments are empty by default, and unfortunately, I don't have access to the `argv` within the class to find the global arguments parsed through command line or the launch. Is there a way to access what ros arguments are parsed from within ros2 node? If so, how?

I need this because I have set the use_global_arguments options to false for the other nodes I create inside my process to not affect by global arguments, but however, I would like them to share the arguments of remapping of topics services etc. For this reason, it would be interesting to have access to the argv and argc from within the ROS2 node

Thank you in advance

fujitatomoya, 2024-08-23:
As far as i know, global arguments are parsed in the initialization of Context (rcl_init), and stored in `rcl_context_s` below.
and depends on user setting for `use_global_arguments`, global arguments are applied to nodes and endpoints.

https://github.com/ros2/rcl/blob/522f3476c76d911d72922584b18df1fa1761f608/rcl/include/rcl/context.h#L113-L117

`rcl_arguments_impl_s` is `rcl` internal implementation, so `rclcpp` cannot access this structure.

https://github.com/ros2/rcl/blob/522f3476c76d911d72922584b18df1fa1761f608/rcl/src/rcl/arguments_impl.h#L29

I think we need to have something like `Context::get_remap_rules()` to return the remap rules from the global arguments parsed in Context.
I am not even sure why `rcl_arguments_impl_s` is implemented as internal structure... probably concealment for maintenance?...

saikishor, 2024-08-23:
@fujitatomoya Thank you so much for responding.

Yes, having a function like `get_remap_rules` would be nice to have this information. I see that even the `rcl_context_impl_s` https://github.com/ros2/rcl/blob/522f3476c76d911d72922584b18df1fa1761f608/rcl/src/rcl/context_impl.h#L29-L41 is also an internal structure. I don't know the reason for it either. 

A method like `Context::get_remap_rules()` as you proposed will need a new method in the rcl library to add it to context. I'm worried that if this would be open to backportable till Humble

sven-hoek, 2025-07-03:
If you just want to have a different node name (and possibly namespace), while keeping the other args (including remappings), this worked for me:
```cpp
const std::vector<std::string> arguments = {"-r", "__name:=" + node_name, "-r", "__ns:=" + node_namespace};
const auto node_options = rclcpp::NodeOptions().use_global_arguments(true).arguments(arguments);
m_node = rclcpp::Node::make_shared(ros_node_name, node_options);
```

Not sure if and how it works to overwrite or even delete other arguments but it worked for my use case and might help others looking for a solution.