ros2/rclcpp #2613: :farmer: TestServiceIntrospection.service_introspection_enable_disable_events failing in Rolling, Jazzy and Iron Connext

Crola1702, 2024-08-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Noble
- Installation type:
  - Source
- Version or commit hash:
  - Rolling
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - Rclcpp

#### Steps to reproduce issue
1. Run a build in [Rci__nightly-connext_ubuntu_noble_amd64](https://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/)
2. See test regression fail

#### Additional information

Reference build:
* [projectroot.test.rclcpp.test_service_introspection](https://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/124/testReport/junit/projectroot.test/rclcpp/test_service_introspection/)
* [rclcpp.TestServiceIntrospection.service_introspection_enable_disable_events](https://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/124/testReport/junit/rclcpp/TestServiceIntrospection/service_introspection_enable_disable_events/)

Log output:
```
[ RUN      ] TestServiceIntrospection.service_introspection_enable_disable_events
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
/tmp/ws/src/ros2/rclcpp/rclcpp/test/rclcpp/test_service_introspection.cpp:201: Failure
Expected equality of these values:
  events.size()
    Which is: 1
  2U
    Which is: 2

/tmp/ws/src/ros2/rclcpp/rclcpp/test/rclcpp/test_service_introspection.cpp:218: Failure
Expected equality of these values:
  events.size()
    Which is: 3
  4U
    Which is: 4

[  FAILED  ] TestServiceIntrospection.service_introspection_enable_disable_events (3309 ms)
```

It's a flaky with a high percentage failure rate:
job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
Ici__nightly-connext_ubuntu_jammy_amd64|2024-08-28|2024-08-14|6|4|66.67
Rci__nightly-connext_ubuntu_noble_amd64|2024-08-27|2024-08-14|12|11|91.67
Jci__nightly-connext_ubuntu_noble_amd64|2024-08-26|2024-08-13|6|5|83.33

fujitatomoya, 2024-10-15:
@fgallegosalido this test also fails with rmw_connextdds only, can you take a look at my comment on https://github.com/ros2/rclcpp/pull/2651? i was trying to address this unstable problem, but it does not work as expected.

fgallegosalido, 2024-10-16:
I will look into it and let you know what I find.

fujitatomoya, 2024-10-25:
@fgallegosalido just checking, any updates on this?

fgallegosalido, 2024-10-28:
@fujitatomoya sorry, not much was done. I was able to reproduce the issue and started working on it, but couldn't find the root cause, so I moved to do other things. I will resume the work on it as soon as possible

fujitatomoya, 2024-10-29:
@fgallegosalido no worries, thanks for working on this. i will revisit https://github.com/ros2/rclcpp/pull/2651 to see if i can find some clue on this.

PR #2684 (2024-11-30): Omnibus fixes for running tests with Connext.
When running the tests with RTI Connext as the default RMW, some of the tests are failing.  There are three different failures fixed here:

1.  Setting the liveliness duration to a value smaller than a microsecond causes Connext to throw an error.  Set it to a millisecond.

2.  Using the SystemDefaultsQoS sets the QoS to KEEP_LAST 1. Connext is somewhat slow in this regard, so it can be the case that we are overwriting a previous service introspection event with the next one.  Switch to the ServicesDefaultQoS in the test, which ensures we will not  [... cut]
PR #2690 (2024-12-03): Omnibus fixes for running tests with Connext. (backport #2684)
When running the tests with RTI Connext as the default RMW, some of the tests are failing.  There are three different failures fixed here:

1.  Setting the liveliness duration to a value smaller than a microsecond causes Connext to throw an error.  Set it to a millisecond.

2.  Using the SystemDefaultsQoS sets the QoS to KEEP_LAST 1. Connext is somewhat slow in this regard, so it can be the case that we are overwriting a previous service introspection event with the next one.  Switch to the ServicesDefaultQoS in the test, which ens [... cut]
closed 2024-11-30 by fujitatomoya