ros2/rclcpp #2637: From humble the C++ class loader takes more time to load the libraries

prabeenkumar122, 2024-10-01:
I am using class-loader in C++ to load the libraries at run-time. From humble the class loader is taking more time( 2x-3x more ) to load the libs in mac platform( macOS 14 ).

- Operating System:
  - MAC OS 14
- Installation type:
  - Latest Humble built from source
- DDS implementation:
  - Fast RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Load a library using C++ class loader in mac (I found in mac OS 14) platform and see the time taken using humble. Same behavior is also seen in Jazzy.

fujitatomoya, 2024-10-01:
@clalancette i think we can move this to https://github.com/ros/class_loader?

@prabeenkumar122 

> class loader is taking more time( 2x-3x more )

against what platform or environment you are comparing this? and it would be helpful if you share numbers as comparison.

clalancette, 2024-10-01:
> @clalancette i think we can move this to https://github.com/ros/class_loader?

Yes, but unfortunately we can't transfer issues between organizations.  So I'll close this one out; @prabeenkumar122 please feel free to reopen on https://github.com/ros/class_loader .

closed 2024-10-01 by clalancette