ros2/rclcpp #2641: Humble - Ubuntu 20.04 - Link Error

Ryanf55, 2024-10-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal
- Installation type:
  - Source
- Version or commit hash:
**humble_min.repos**
```
repositories:
  ament/ament_cmake:
    type: git
    url: https://github.com/ament/ament_cmake.git
    version: humble
  ament/ament_index:
    type: git
    url: https://github.com/ament/ament_index.git
    version: humble
  ament/ament_lint:
    type: git
    url: https://github.com/ament/ament_lint.git
    version: humble
  ament/ament_package:
    type: git
    url: https://github.com/ament/ament_package.git
    version: humble
  build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext:
    type: git
    url: https://github.com/foonathan/memory.git
    version: 0f0775770fd1c506fa9c5ad566bd6ba59659db66
  build/libyaml_vendor/libyaml-0.2.5-prefix/src/libyaml-0.2.5:
    type: git
    url: https://github.com/yaml/libyaml.git
    version: 2c891fc7a770e8ba2fec34fc6b545c672beb37e6
  build/mcap_vendor/_deps/lz4-src:
    type: git
    url: https://github.com/lz4/lz4.git
    version: d44371841a2f1728a3f36839fd4b7e872d0927d3
  build/zstd_vendor/zstd-1.4.4-prefix/src/zstd-1.4.4:
    type: git
    url: https://github.com/facebook/zstd.git
    version: 10f0e6993f9d2f682da6d04aa2385b7d53cbb4ee
  eProsima/Fast-CDR:
    type: git
    url: https://github.com/eProsima/Fast-CDR.git
    version: da2987299ee3104bb0393cf0afc8aad6fb848dc1
  eProsima/Fast-DDS:
    type: git
    url: https://github.com/eProsima/Fast-DDS.git
    version: 2.6.x
  eProsima/foonathan_memory_vendor:
    type: git
    url: https://github.com/eProsima/foonathan_memory_vendor.git
    version: master
  ros-tooling/libstatistics_collector:
    type: git
    url: https://github.com/ros-tooling/libstatistics_collector.git
    version: humble
  ros/class_loader:
    type: git
    url: https://github.com/ros/class_loader.git
    version: humble
  ros/pluginlib:
    type: git
    url: https://github.com/ros/pluginlib.git
    version: humble
  ros2/ament_cmake_ros:
    type: git
    url: https://github.com/ros2/ament_cmake_ros.git
    version: humble
  ros2/common_interfaces:
    type: git
    url: https://github.com/ros2/common_interfaces.git
    version: humble
  ros2/console_bridge_vendor:
    type: git
    url: https://github.com/ros2/console_bridge_vendor.git
    version: humble
  ros2/libyaml_vendor:
    type: git
    url: https://github.com/ros2/libyaml_vendor.git
    version: humble
  ros2/rcl:
    type: git
    url: https://github.com/ros2/rcl.git
    version: humble
  ros2/rcl_interfaces:
    type: git
    url: https://github.com/ros2/rcl_interfaces.git
    version: humble
  ros2/rcl_logging:
    type: git
    url: https://github.com/ros2/rcl_logging.git
    version: humble
  ros2/rclcpp:
    type: git
    url: https://github.com/ros2/rclcpp.git
    version: humble
  ros2/rcpputils:
    type: git
    url: https://github.com/ros2/rcpputils.git
    version: humble
  ros2/rcutils:
    type: git
    url: https://github.com/ros2/rcutils.git
    version: humble
  ros2/rmw:
    type: git
    url: https://github.com/ros2/rmw.git
    version: humble
  ros2/rmw_dds_common:
    type: git
    url: https://github.com/ros2/rmw_dds_common.git
    version: humble
  ros2/rmw_fastrtps:
    type: git
    url: https://github.com/ros2/rmw_fastrtps.git
    version: humble
  ros2/rmw_implementation:
    type: git
    url: https://github.com/ros2/rmw_implementation.git
    version: humble
  ros2/ros2_tracing:
    type: git
    url: https://github.com/ros2/ros2_tracing.git
    version: humble
  ros2/rosbag2:
    type: git
    url: https://github.com/ros2/rosbag2.git
    version: humble
  ros2/rosidl:
    type: git
    url: https://github.com/ros2/rosidl.git
    version: humble
  ros2/rosidl_defaults:
    type: git
    url: https://github.com/ros2/rosidl_defaults.git
    version: humble
  ros2/rosidl_typesupport:
    type: git
    url: https://github.com/ros2/rosidl_typesupport.git
    version: humble
  ros2/rosidl_typesupport_fastrtps:
    type: git
    url: https://github.com/ros2/rosidl_typesupport_fastrtps.git
    version: humble
  ros2/spdlog_vendor:
    type: git
    url: https://github.com/ros2/spdlog_vendor.git
    version: humble
  ros2/test_interface_files:
    type: git
    url: https://github.com/ros2/test_interface_files.git
    version: humble
  ros2/tinyxml2_vendor:
    type: git
    url: https://github.com/ros2/tinyxml2_vendor.git
    version: humble
  ros2/unique_identifier_msgs:
    type: git
    url: https://github.com/ros2/unique_identifier_msgs.git
    version: humble
  ros2/yaml_cpp_vendor:
    type: git
    url: https://github.com/ros2/yaml_cpp_vendor.git
    version: humble
 ```
- DDS implementation:
  - fastrtps
- Client library (if applicable):
  - rclcpp_components

#### Steps to reproduce issue
```bash
vcs import --input humble_min.repos
colcon build --cmake-args -DBUILD_TESTING=OFF
```

Logs:
```
Finished <<< rclcpp [0.48s]                       
Starting >>> rosbag2_test_common
Starting >>> rclcpp_lifecycle
Starting >>> rclcpp_action
Starting >>> rclcpp_components
Finished <<< rosbag2_test_common [0.21s]                                                                                                       
Starting >>> rosbag2_storage
Finished <<< rclcpp_lifecycle [0.27s]                                                                                                                                
Starting >>> test_tracetools
Finished <<< rclcpp_action [0.28s]
--- stderr: rclcpp_components  

--- stderr: rclcpp_components                                                                                                             
/usr/bin/ld: /ws/ubuntu20.04/install/rcl/lib/librcl.so: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__rcl_interfaces__msg__Log'

...
install/rclcpp/lib/librclcpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > >()'

```

#### Expected behavior

A configure error if there is a missing dependency

#### Actual behavior

A link error, I suspect from not properly calling find_package before linking a target.

#### Additional information

fujitatomoya, 2024-10-04:
@Ryanf55 you might already know that Ubuntu Focal is Tier 3 platform for Humble, https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027.

Ryanf55, 2024-10-04:
> @Ryanf55 you might already know that Ubuntu Focal is Tier 3 platform for Humble, https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027.

Yep. It's on hardware that can't be updated to the tier one OS.