ros2/rclcpp #2652: Waitset not triggering timer when initialized in lifecycle node

luca-della-vedova, 2024-10-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - Jazzy binaries (but reproducible from source, bisected to the commit linked below which introduced the issue)
- Version or commit hash:
  -  I bisected and found the exact commit that introduced this behavior (it was OK before): https://github.com/ros2/rclcpp/commit/5632a09af212c624c24c3820c765ac4bab5a8ddc
- DDS implementation:
  - Cyclone
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

* Build my "minimal" reproducible repo https://github.com/luca-della-vedova/timer_repro.
* Run the node
* In another terminal (in this order!):

```
ros2 lifecycle set /minimal_timer configure
ros2 lifecycle set /minimal_timer activate
ros2 run examples_rclcpp_minimal_service service_main
```

#### Expected behavior

Since the service callback destroys only `timer_`, `timer2_` should keep running and `Hello, world!` should keep printing.

#### Actual behavior

All timers are stopped and no `Hello, world!` is printed.

#### Additional information

The following might shed some light into this.

* If the timer that is erronously not triggered is initialized in the constructor, rather than in a lifecycle transition, the issue does not arise.
* If the action server is ran before the lifecycle transition (which means `timer2_` is only created after `timer_` is destroyed) the issue does not arise.
* If I manually add a `|| true` [here](https://github.com/ros2/rclcpp/blob/0be8aa013e4ed94bcab88527887e85130816d256/rclcpp/src/rclcpp/executor.cpp#L734) making this condition always evaluate to true, all timers are executed, however I added some printing when timers are cleaned up / added and I can see that every iteration a timer gets added and removed, so something odd is going on there.

So I suspect there is something about how timers (and maybe other entities?) are added to wait sets inside lifecycle nodes that creates this issue.

mjcarroll, 2024-10-16:
This definitely sounds like a bug.  

> If I manually add a || true [here](https://github.com/ros2/rclcpp/blob/0be8aa013e4ed94bcab88527887e85130816d256/rclcpp/src/rclcpp/executor.cpp#L734) making this condition always evaluate to true,

This is going to force the waitset to always be rebuilt, which is not good for performance.  It's likely that the bug is somewhere else (in how timers are added?)

I probably won't get a chance to look at this until after roscon, but thank you for the full repro example and detailed steps.

mjcarroll, 2024-10-16:
> lifecycle nodes do not do anything special here

My assumption as well is that something about the execution order from lifecycle nodes is surfacing this, but it's not something that would be unique to lifecycle nodes.

fmrico, 2024-10-17:
Hi,

I discovered this bug five days ago at https://github.com/fmrico/cascade_lifecycle/pull/14, and now I am solving a problem that is directly related at PlanSys2 (https://github.com/ros/rosdistro/pull/43212#issuecomment-2418582680). I am solving this now by changing the SingleThreadedExecutors to EventsExecutors.

Basically, if I have two nodes in an executor, each one with a timer, if I destroy one of the nodes, the timer callback in the other node is no longer called. 

I can investigate more if you want.

jmachowinski, 2024-12-13:
I found the bug:
https://github.com/ros2/rclcpp/blob/8c0161a07f1a44db023650d10f1b2e581c64e1f1/rclcpp/include/rclcpp/wait_result.hpp#L176-L178

We got a mismatch here between the sizeof the rcl waitset and the rclcpp waitset. Therefore the ready timer in the rcl waitset is never checked. The fix is simple, but it is in exposed template code...

@clalancette Are we breaking ABI if we are patching the templates ?

jmachowinski, 2024-12-13:
My initial analysis was not correct. We got a mismatch between the index in the rcl_waitset and the index in the rclcpp waitset. Thats the bug.

clalancette, 2024-12-13:
> @clalancette Are we breaking ABI if we are patching the templates ?

It depends on what we are changing, I think.  What I'm going to say here is that we open the PR to fix things against `rolling`, and get that in.  After that we can figure out what we need to do to backport it.

anders-clement-tt, 2025-01-21:
We have also stumbled upon this issue. Therefore I will add our setup and findings, in case it is helpful.

**Required Info:**
Operating System:
Ubuntu 24.04
Installation type:
Jazzy binaries as of january 21st 2025.

DDS implementation:
Cyclone, fastrtps, zenoh
Client library (if applicable):
rclcpp

**Minimal example**
We have tested with the following minimal example:
```cpp
#include "rclcpp/rclcpp.hpp"

class TestNode : public rclcpp::Node
{
public:
    TestNode() : Node("test_node")
    {
        m_test_timer = create_timer(std::chrono::milliseconds(110), [this]() { 
           RCLCPP_INFO(get_logger(), "Test timer 1 callback.");
        });

        m_test_timer2 = create_timer(std::chrono::milliseconds(500), [this]() {
            RCLCPP_INFO(get_logger(), "Test 2 timer callback.");
            m_test_timer2.reset();
        });
    }

    rclcpp::TimerBase::SharedPtr m_test_timer;
    rclcpp::TimerBase::SharedPtr m_test_timer2;
};

int main(int argc, char** argv)
{
    rclcpp::init(argc, argv);
    auto test_node = std::make_shared<TestNode>();
    rclcpp::spin(test_node);
    rclcpp::shutdown();
    return 0;
}
```

**Expected behavior:**
Timer 1 will fire every 110 ms, and timer 2 will fire once after 500 ms. After timer 2 has fired, timer 1 is expected to keep firing.

**Results:**
Using ```rclcpp::spin(test_node);``` Thus using the SingleThreadedExecutor, timer 1 will keep firing only timer 2 is reset, after which point it will stop firing: 

```
~$ ros2 run test_package test_node
[INFO] [1737445725.640752999] [test_node]: Test timer 1 callback.
[INFO] [1737445725.750123938] [test_node]: Test timer 1 callback.
[INFO] [1737445725.860374645] [test_node]: Test timer 1 callback.
[INFO] [1737445725.970357124] [test_node]: Test timer 1 callback.
[INFO] [1737445726.030081623] [test_node]: Test 2 timer callback.
```

**Workarounds:**
Changing to using either the MultiThreadedExecutor or the experimental EventsExecutor seems to resolve the problem (We observe expected behavior where timer 1 keeps firing after timer 2 is reset).

closed 2025-02-11 by fujitatomoya