ros2/rclcpp #2666: Segmentation fault with global Node

allbluelai, 2024-11-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 + VMware
- Installation type:
add [backward_ros](https://github.com/pal-robotics/backward_ros) to source and build the source
  ```sh
    colcon build --merge-install --packages-skip-build-finished --cmake-args -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Debug
  ```
- Version or commit hash:
  - humble
- DDS implementation:
  - rmw_fastrtps_cpp and rmw_cyclonedds_cpp
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
```cpp
// global.node
#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"

rclcpp::Node::SharedPtr g_node = nullptr;

int main(int argc, char *argv[]) {
  rclcpp::init(argc, argv);
  g_node = rclcpp::Node::make_shared("global_node");
  rclcpp::spin(g_node);
  rclcpp::shutdown();
  return 0;
}
```

```cpp
// local_node.cpp
#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"

int main(int argc, char *argv[]) {
  rclcpp::init(argc, argv);
  rclcpp::Node::SharedPtr l_node = rclcpp::Node::make_shared("local_node");
  rclcpp::spin(l_node);
  rclcpp::shutdown();
  return 0;
}
```

#### Expected behavior
Exit without `Segmentation fault` when Ctrl+C
```sh
^C[INFO] [1731149893.815000507] [rclcpp]: signal_handler(signum=2)
```

#### Actual behavior
+ rmw_cyclonedds_cpp+global_node 
+ rmw_cyclonedds_cpp+local_node 
+ rmw_fastrtps_cpp+local_node 
All three above work fine:
```sh
$ RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run cpp_pubsub global_node 
^C[INFO] [1731149893.815000507] [rclcpp]: signal_handler(signum=2)
```

```sh
$ RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run cpp_pubsub local_node 
^C[INFO] [1731150107.206215360] [rclcpp]: signal_handler(signum=2)
```

[... 233 lines cut ...]
        499:     return RMW_RET_INCORRECT_RMW_IMPLEMENTATION);
        500: 
      > 501:   return rmw_fastrtps_shared_cpp::__rmw_destroy_service(
        502:     eprosima_fastrtps_identifier, node, service);
        503: }
#2    Source "/home/john/ros2_humble/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_service.cpp", line 131, in __rmw_destroy_service [0x72ccb6ba567d]
        128:     }
        129: 
        130:     // Delete topics and unregister types
      > 131:     remove_topic_and_type(participant_info, request_topic, info->request_type_support_);
        132:     remove_topic_and_type(participant_info, response_topic, info->response_type_support_);
        133: 
        134:     // Delete CustomServiceInfo structure
#1    Source "/home/john/ros2_humble/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/utils.cpp", line 124, in remove_topic_and_type [0x72ccb6bc8e8e]
        121:   // TODO(MiguelCompany): We only create Topic instances at the moment, but this may
        122:   // change in the future if we start supporting other kinds of TopicDescription
        123:   // (like ContentFilteredTopic)
      > 124:   auto topic = dynamic_cast<const eprosima::fastdds::dds::Topic *>(topic_desc);
        125: 
        126:   if (nullptr != topic) {
        127:     participant_info->participant_->delete_topic(topic);
#0    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x72ccb82acc9d, in __dynamic_cast
Segmentation fault (Signal sent by the kernel [(nil)])
[ros2run]: Segmentation fault
```

MichaelOrlov, 2024-11-21:
@allbluelai The root cause of the segfault is likely that the domain participant was already destructed when we reached the destructor of the globally defined node.
The problem is that order of destruction for the globally defined objects is in the "grey" area of the C++ and not defined.
A possible workaround would be to try to call the destructor of the globally defined node explicitly somewhere in the main function.

allbluelai, 2024-11-29:
@MichaelOrlov 
I'm sorry for not getting back to you sooner. Yes, explicit destruction of the global Node is required.
I added the following code and no segfault occurred：
```cpp
rclcpp::shutdown();
g_node.reset();
```

closed 2024-11-29 by allbluelai