ros2/rclcpp #2680: Segmentaion fault with test_client_common

YuanYuYuan, 2024-11-20:
## Bug report

**Required Info:**

- Operating System:
  [ros/rolling docker](https://hub.docker.com/layers/library/ros/rolling/images/sha256-9ea1938d30d851fa79df496aeca4d144b2492a1155b7c063e9da80e7f73ebc8e?context=explore) 
- DDS implementation:
  - fastrtps or cyclonedds
- Client library (if applicable):
  - rclcpp: https://github.com/ros2/rclcpp/commit/e854bb29cdb0a3fe8e53c4d683b848124c249cdf

#### Steps to reproduce issue
1. Download rclcpp https://github.com/ros2/rclcpp/commit/e854bb29cdb0a3fe8e53c4d683b848124c249cdf
2. Build rclcpp
3. Run the test with 
```console
➜ RMW_IMPLEMENTATION=rmw_cyclonedds_cpp /workspace/build/rclcpp/test/rclcpp/test_client_common
Running main() from /opt/ros/rolling/src/gtest_vendor/src/gtest_main.cc
[==========] Running 26 tests from 2 test suites.
[----------] Global test environment set-up.
[----------] 13 tests from TestAllClientTypesWithServer/Client, where TypeParam = rclcpp::Client<test_msgs::srv::Empty>
[ RUN      ] TestAllClientTypesWithServer/Client.async_send_request
[       OK ] TestAllClientTypesWithServer/Client.async_send_request (10 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.test_client_remove_pending_request
[       OK ] TestAllClientTypesWithServer/Client.test_client_remove_pending_request (4 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.prune_requests_older_than_no_pruned
[       OK ] TestAllClientTypesWithServer/Client.prune_requests_older_than_no_pruned (3 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.prune_requests_older_than_with_pruned
[       OK ] TestAllClientTypesWithServer/Client.prune_requests_older_than_with_pruned (5 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.async_send_request_rcl_send_request_error
[       OK ] TestAllClientTypesWithServer/Client.async_send_request_rcl_send_request_error (5 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.async_send_request_rcl_service_server_is_available_error
[       OK ] TestAllClientTypesWithServer/Client.async_send_request_rcl_service_server_is_available_error (9 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.take_response
[       OK ] TestAllClientTypesWithServer/Client.take_response (5 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.on_new_response_callback
[       OK ] TestAllClientTypesWithServer/Client.on_new_response_callback (312 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.client_qos
[       OK ] TestAllClientTypesWithServer/Client.client_qos (8 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.rcl_client_request_publisher_get_actual_qos_error
[       OK ] TestAllClientTypesWithServer/Client.rcl_client_request_publisher_get_actual_qos_error (4 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.rcl_client_response_subscription_get_actual_qos_error
[       OK ] TestAllClientTypesWithServer/Client.rcl_client_response_subscription_get_actual_qos_error (7 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.async_send_request_callback_with_request
[       OK ] TestAllClientTypesWithServer/Client.async_send_request_callback_with_request (6 ms)
[ RUN      ] TestAllClientTypesWithServer/Client.qos_depth
[       OK ] TestAllClientTypesWithServer/Client.qos_depth (68 ms)
[----------] 13 tests from TestAllClientTypesWithServer/Client (451 ms total)

[----------] 13 tests from TestAllClientTypesWithServer/GenericClient, where TypeParam = rclcpp::GenericClient
[ RUN      ] TestAllClientTypesWithServer/GenericClient.async_send_request
[       OK ] TestAllClientTypesWithServer/GenericClient.async_send_request (9 ms)
[ RUN      ] TestAllClientTypesWithServer/GenericClient.test_client_remove_pending_request
[       OK ] TestAllClientTypesWithServer/GenericClient.test_client_remove_pending_request (6 ms)
[ RUN      ] TestAllClientTypesWithServer/GenericClient.prune_requests_older_than_no_pruned
[1]    479 segmentation fault (core dumped)  RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

The test fails either with cyclonedds or fastrtps.

- GDB backtrace
```
(gdb) bt
#0  0x00005603b2faf77c in std::_Function_base::~_Function_base (this=0x5603bcf4e558, __in_chrg=<optimized out>) at /usr/include/c++/13/bits/std_function.h:244
#1  0x00005603b3011b80 in std::function<void (std::shared_future<std::shared_ptr<void> >)>::~function() (this=0x5603bcf4e558, __in_chrg=<optimized out>)
    at /usr/include/c++/13/bits/std_function.h:334
#2  0x00005603b3011ba0 in std::_Head_base<0ul, std::function<void (std::shared_future<std::shared_ptr<void> >)>, false>::~_Head_base() (this=0x5603bcf4e558,
    __in_chrg=<optimized out>) at /usr/include/c++/13/tuple:188
#3  0x00005603b3011bc4 in std::_Tuple_impl<0ul, std::function<void (std::shared_future<std::shared_ptr<void> >)>, std::shared_future<std::shared_ptr<void> >, std::promise<std::shared_ptr<void> > >::~_Tuple_impl() (this=0x5603bcf4e530, __in_chrg=<optimized out>) at /usr/include/c++/13/tuple:259
#4  0x00005603b3011bf0 in std::tuple<std::function<void (std::shared_future<std::shared_ptr<void> >)>, std::shared_future<std::shared_ptr<void> >, std::promise<std::shared_ptr<void> > >::~tuple() (this=0x5603bcf4e530, __in_chrg=<optimized out>) at /usr/include/c++/13/tuple:746
[... 36 lines cut ...]
#28 0x00005603b2ff540c in rclcpp::detail::prune_requests_older_than_impl<std::map<long, std::pair<std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::variant<std::promise<std::shared_ptr<void> >, std::tuple<std::function<void (std::shared_future<std::shared_ptr<void> >)>, std::shared_future<std::shared_ptr<void> >, std::promise<std::shared_ptr<void> > > > >, std::less<long>, std::allocator<std::pair<long const, std::pair<std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::variant<std::promise<std::shared_ptr<void> >, std::tuple<std::function<void (std::shared_future<std::shared_ptr<void> >)>, std::shared_future<std::shared_ptr<void> >, std::promise<std::shared_ptr<void> > > > > > > >, std::allocator<long> >(std::map<long, std::pair<std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::variant<std::promise<std::shared_ptr<void> >, std::tuple<std::function<void (std::shared_future<std::shared_ptr<void> >)>, std::shared_future<std::shared_ptr<void> >, std::promise<std::shared_ptr<void> > > > >, std::less<long>, std::allocator<std::pair<long const, std::pair<std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::variant<std::promise<std::shared_ptr<void> >, std::tuple<std::function<void (std::shared_future<std::shared_ptr<void> >)>, std::shared_future<std::shared_ptr<void> >, std::promise<std::shared_ptr<void> > > > > > > >&, std::mutex&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::vector<long, std::allocator<long> >*) (pending_requests=std::map with 1 element = {...}, pending_requests_mutex=...,
    time_point=std::chrono::sys_time = { 1732111090907666181ns [2024-11-20 13:58:10] }, pruned_requests=0x0) at /workspace/src/rclcpp/rclcpp/include/rclcpp/client.hpp:126
#29 0x00005603b2ff0515 in rclcpp::GenericClient::prune_requests_older_than<std::allocator<long> > (this=0x5603bd0f1a20,
    time_point=std::chrono::sys_time = { 1732111090907666181ns [2024-11-20 13:58:10] }, pruned_requests=0x0)
    at /workspace/src/rclcpp/rclcpp/include/rclcpp/generic_client.hpp:201
#30 0x00005603b2fe00cd in TestAllClientTypesWithServer_prune_requests_older_than_no_pruned_Test<rclcpp::GenericClient>::TestBody (this=0x5603bd06cd40)
    at /workspace/src/rclcpp/rclcpp/test/rclcpp/test_client_common.cpp:236
#31 0x00005603b3057c81 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (object=0x5603bd06cd40, method=&virtual testing::Test::TestBody(),
    location=0x5603b3078bfb "the test body") at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2612
#32 0x00005603b304fd51 in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=0x5603bd06cd40, method=&virtual testing::Test::TestBody(),
    location=0x5603b3078bfb "the test body") at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2648
#33 0x00005603b302805e in testing::Test::Run (this=0x5603bd06cd40) at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2687
[... cut ...]
    location=0x5603b3079648 "auxiliary test code (environments or event listeners)") at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:2648
#39 0x00005603b3038c5f in testing::UnitTest::Run (this=0x5603b30ce700 <testing::UnitTest::GetInstance()::instance>) at /opt/ros/rolling/src/gtest_vendor/./src/gtest.cc:5484
#40 0x00005603b3013459 in RUN_ALL_TESTS () at /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:2317
#41 0x00005603b3013441 in main (argc=1, argv=0x7ffd46461e98) at /opt/ros/rolling/src/gtest_vendor/src/gtest_main.cc:64
```

clalancette, 2024-11-20:
I'm not sure, but this may be a case where we just have to do releases into Rolling.  The reason I think that is that in our from-source builds last night at https://ci.ros2.org/view/nightly/ , we didn't see this failure.  I'll leave this open for now and do a round of Rolling releases in the next couple of days.

clalancette, 2024-11-20:
I just tried rebuilding again locally, and I'm unable to reproduce this with any of `rmw_fastrtps_cpp`, `rmw_cyclonedds_cpp`, or `rmw_connextdds`.

However, I am able to reproduce this *sometimes* with `rmw_zenoh_cpp`, on both the `rolling` branch and the `dev/1.0.0` branch.  This needs further investigation.

clalancette, 2024-11-20:
All right, this is a bug in `rmw_zenoh_cpp`.  See https://github.com/ros2/rmw_zenoh/pull/320, which should fix it, at least on the `rolling` branch.

YuanYuYuan, 2024-12-04:
Hi @clalancette! Can you help transfer this issue back to ros2/rmw_zenoh? I found that this one is router-status-dependent. 

Procedure to reproduce it:

1. Launch zenohd.
2. Run rclcpp/test_executors or rclcpp/test_executor until we encounter an error.
3. Run rclcpp/test_client_common. The test can fail consistently in this way.