ros2/rclcpp #2681: [Test failure] test_service_introspection client_gid mismatch

YuanYuYuan, 2024-11-21:
## Bug report

**Required Info:**

- Operating System:
  [ros/rolling docker](https://hub.docker.com/layers/library/ros/rolling/images/sha256-9ea1938d30d851fa79df496aeca4d144b2492a1155b7c063e9da80e7f73ebc8e?context=explore) 
- DDS implementation:
  - rmw_fastrtps_cpp: failed
  - rmw_cyclonedds_cpp: passed
  - rmw_zenoh_cpp 
    - rolling(https://github.com/ros2/rmw_zenoh/commit/8306a63313cd9e21883344deff9abc782636a464): passed
    - 1.0(https://github.com/ZettaScaleLabs/rmw_zenoh/commit/2e3ad436aabf411e4463f169b444714e40af6d67): failed
- Client library (if applicable):
  - rclcpp: https://github.com/ros2/rclcpp/commit/64dd644469d7fb6f2ed64e609e5bd48b6b23f4ce
  - rcl: https://github.com/ros2/rcl

#### Steps to reproduce issue

```console
➜ colcon test --event-handlers console_cohesion+ console_direct+ --packages-select rclcpp --ctest-args -R test_service_introspection
Starting >>> rclcpp
UpdateCTestConfiguration  from :/workspace/build/rclcpp/CTestConfiguration.ini
Parse Config file:/workspace/build/rclcpp/CTestConfiguration.ini
   Site: ZenS13
   Build name: (empty)
 Add coverage exclude regular expressions.
Create new tag: 20241121-1525 - Experimental
UpdateCTestConfiguration  from :/workspace/build/rclcpp/CTestConfiguration.ini
Parse Config file:/workspace/build/rclcpp/CTestConfiguration.ini
Test project /workspace/build/rclcpp
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 74
    Start 74: test_service_introspection

74: Test command: /usr/bin/python3 "-u" "/opt/ros/rolling/share/ament_cmake_test/cmake/run_test.py" "/workspace/build/rclcpp/test_results/rclcpp/test_service_introspection.gtest.xml" "--package-name" "rclcpp" "--output-file" "/workspace/build/rclcpp/ament_cmake_gmock/test_service_introspection.txt" "--command" "/workspace/build/rclcpp/test/rclcpp/test_service_introspection" "--gtest_output=xml:/workspace/build/rclcpp/test_results/rclcpp/test_service_introspection.gtest.xml"
74: Working Directory: /workspace/build/rclcpp/test/rclcpp
74: Test timeout computed to be: 60
74: -- run_test.py: invoking following command in '/workspace/build/rclcpp/test/rclcpp':
74:  - /workspace/build/rclcpp/test/rclcpp/test_service_introspection --gtest_output=xml:/workspace/build/rclcpp/test_results/rclcpp/test_service_introspection.gtest.xml
74: Running main() from gmock_main.cc
74: [==========] Running 3 tests from 1 test suite.
74: [----------] Global test environment set-up.
74: [----------] 3 tests from TestServiceIntrospection
74: [ RUN      ] TestServiceIntrospection.service_introspection_nominal
74: /workspace/src/rclcpp/rclcpp/test/rclcpp/test_service_introspection.cpp:131: Failure
74: Value of: client_gid_arr
74: Expected: is equal to { '\x1' (1), '\xF' (15), 'h' (104, 0x68), '\xD9' (217), '\\' (92, 0x5C), 'p' (112, 0x70), '\\' (92, 0x5C), '\v' (11, 0xB), '\0', '\0', '\0', '\0', '\0', '\0', '\x14' (20), '\x4' (4) }
74:   Actual: { '\x1' (1), '\xF' (15), 'h' (104, 0x68), '\xD9' (217), '\\' (92, 0x5C), 'p' (112, 0x70), '\\' (92, 0x5C), '\v' (11, 0xB), '\0', '\0', '\0', '\0', '\0', '\0', '\x15' (21), '\x3' (3) }
74:
74: [  FAILED  ] TestServiceIntrospection.service_introspection_nominal (24 ms)
74: [ RUN      ] TestServiceIntrospection.service_introspection_enable_disable_events
74: [       OK ] TestServiceIntrospection.service_introspection_enable_disable_events (1009 ms)
74: [ RUN      ] TestServiceIntrospection.service_introspection_enable_disable_event_content
74: [       OK ] TestServiceIntrospection.service_introspection_enable_disable_event_content (12 ms)
74: [----------] 3 tests from TestServiceIntrospection (1045 ms total)
74:
74: [----------] Global test environment tear-down
74: [==========] 3 tests from 1 test suite ran. (1046 ms total)
74: [  PASSED  ] 2 tests.
74: [  FAILED  ] 1 test, listed below:
74: [  FAILED  ] TestServiceIntrospection.service_introspection_nominal
74:
74:  1 FAILED TEST
74: -- run_test.py: return code 1
74: -- run_test.py: inject classname prefix into gtest result file '/workspace/build/rclcpp/test_results/rclcpp/test_service_introspection.gtest.xml'
74: -- run_test.py: verify result file '/workspace/build/rclcpp/test_results/rclcpp/test_service_introspection.gtest.xml'
1/1 Test #74: test_service_introspection .......***Failed    1.24 sec

0% tests passed, 1 tests failed out of 1

Label Time Summary:
gmock     =   1.24 sec*proc (1 test)
mimick    =   1.24 sec*proc (1 test)

Total Test time (real) =   1.25 sec

The following tests FAILED:
         74 - test_service_introspection (Failed)
```

clalancette, 2024-11-21:
So we don't see this on the nightlies: https://ci.ros2.org/view/nightly/job/nightly_linux_release/3270/ .  There must be something different about your environment here, I'm not sure what.

YuanYuYuan, 2024-11-29:
I suddenly can't reproduce it as well. Let's close it.