ros2/rclcpp #2688: Minor suggestion for `rclcpp::Node::now()` documentation string

roncapat, 2024-11-28:
https://github.com/ros2/rclcpp/blob/e64627004f4265560addf3532510a83602be5aef/rclcpp/include/rclcpp/node.hpp#L1433-L1439

Notice how "clock_type" is never referenced in the `node.hpp` file.
I believe this is a copy-paste from `rclcpp::Clock::now()` method docstring.

Maybe it's better to say something like:
```
Returns current time from the node time source. If `use_sim_time` parameter is set to `true` while constructing the `Node`, the simulation time is returned, otherwise system time is returned.
```

christophfroehlich, 2024-12-01:
> ```
> Returns current time from the node time source.
> If `use_sim_time` parameter is set to `true` while
> constructing the `Node`, the simulation time is returned,
> otherwise system time is returned.
> ```

I think it doesn't need to be set before construction but can be set afterwards like in [gz_ros2_control](https://github.com/ros-controls/gz_ros2_control/blob/07a2d3b2519e36e3b86194b286869f292e4e6782/gz_ros2_control/src/gz_ros2_control_plugin.cpp#L454-L456)

roncapat, 2024-12-02:
Refined proposal: 

```
Returns current time from the node time source.
If `use_sim_time` parameter is set to `true`
the simulation time is returned,
otherwise system time is returned.
```

fujitatomoya, 2024-12-02:
@roncapat thanks for the suggestion. can you create the PR for this? i am happy to review that.

and one comment for the PR, `LifecycleNode::now()` should be updated accordingly.

closed 2024-12-04 by fujitatomoya