ros2/rclcpp #2705: SystemDefaultsQoS - force to have depth > 0 for Publishers

SteveMacenski, 2024-12-13:
## Feature request

#### Feature description

For SystemDefaultsQoS to work out of the box for IPC, it needs to always strictly have a depth setting > 0. When used in a subscription out of the box, this works fine. For example:

```
  footprint_sub_ = create_subscription<geometry_msgs::msg::Polygon>(
    "footprint",
    nav2_util::DefaultQoS(),
    std::bind(&Costmap2DROS::setRobotFootprintPolygon, this, std::placeholders::_1));
```

However, when used in a publisher, it fails. For example:

```
  footprint_pub_ = create_publisher<geometry_msgs::msg::PolygonStamped>(
    "published_footprint", rclcpp::SystemDefaultsQoS());
```

or 

```
  state_pub_ = this->create_publisher<nav2_msgs::msg::CollisionDetectorState>(
    "collision_detector_state", rclcpp::SystemDefaultsQoS());
```

Thus, I feel that we should update the default for SystemDefaultsQoS for publishers (or in general) to be `1` if not already `> 1` so that folks migrating or implementing IPC don't run into a cacophony of issues, one for each publisher.

fujitatomoya, 2024-12-13:
@SteveMacenski thanks for detailed description, that really helped.

i believe this is a bug for the intra-process publisher, https://github.com/ros2/rclcpp/pull/2707 should fix this.

SteveMacenski, 2024-12-14:
I'll openly admit the use of `qos` vs `get_actual_qos()` is a subtlety I cannot contribute much to the discussion on :-) 

I will say that in your test, I see it with `use_intra_process_comms(true)` without messing with the publisher-specific IPC settings like you do in your current test. I think these are functionally the same, but if you wanted to verify it fixed what I was seeing, a test might be simply:

```
TEST_F(TestPublisher, test_publisher_with_system_default_qos) {
  initialize(rclcpp::NodeOptions().use_intra_process_comms(true));
  using test_msgs::msg::Empty;
  ASSERT_NO_THROW(
  {
    auto publisher = node->create_publisher<Empty>("topic", rclcpp::SystemDefaultsQoS());
  });
}
```

If you see that fail without your change and pass with it, then it fixes the issue case I see :smile:

closed 2024-12-16 by fujitatomoya