ros2/rclcpp #2731: 🧑‍🌾 `rcl_action_server_init2` not declared in rclcpp_action

Crola1702, 2025-01-31:
## Bug Report

Operating System
* Ubuntu Noble
Version:
* Rolling

Reference build:
* https://build.ros2.org/view/Rci/job/Rci__benchmark_ubuntu_noble_amd64/199/

```
12:26:39 /tmp/ws2/src/ros2/rclcpp/rclcpp_action/src/server.cpp:157:19: error: ‘rcl_action_server_init2’ was not declared in this scope; did you mean ‘rcl_action_server_init’?
12:26:39   157 |   rcl_ret_t ret = rcl_action_server_init2(
12:26:39       |                   ^~~~~~~~~~~~~~~~~~~~~~~
12:26:39       |                   rcl_action_server_init
12:26:39 gmake[2]: *** [CMakeFiles/rclcpp_action.dir/build.make:104: CMakeFiles/rclcpp_action.dir/src/server.cpp.o] Error 1
12:26:39 gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/rclcpp_action.dir/all] Error 2
12:26:39 gmake: *** [Makefile:146: all] Error 2
12:26:39 ---
12:26:39 Failed   <<< rclcpp_action [11.5s, exited with code 2]
```

This was introduced in #2699

fujitatomoya, 2025-01-31:
@Crola1702 

CC: @ahcorde @jmachowinski 

this is weird, dependent rcl change has been merged at the same time, https://github.com/ros2/rcl/pull/1201.
is that possible that nightly build just started when exactly those changes were being merged into rolilng branches??? according to the timestamp, CI started after all these changes are merged for me....

can we wait it out to see if what is gonna happen today's nightly?

i tried https://github.com/ros2/ros2/commit/086c0fd24ad51937b87ea99f2276e52541a103f2 and 

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# colcon build --symlink-install --packages-up-to rclcpp
```

and successfully built the `rclcpp`.

clalancette, 2025-01-31:
I think it helps to understand a bit how the Rci__benchmark job gets run.  In particular, what happens is that Rci__nightly-extra-rmw-release job is run first.  After that, Rci__benchmark is run with that extra-rmw-release job as an underlay, and then a bunch of packages are rebuilt on top of that.

So it is possible that either the underlay got built, then the patches were merged, then the overlay was built, or it is also possible that we got very unlucky and the job run in between when one PR was merged and another.  What I personally would do here is wait until the next time this is run and see if it is fixed; if not, then we can look into it deeper.

Crola1702, 2025-02-03:
> What I personally would do here is wait until the next time this is run and see if it is fixed; if not, then we can look into it deeper.

You are right. Last 3 builds of [Rci_benchmark passed](https://build.ros2.org/view/Rci/job/Rci__benchmark_ubuntu_noble_amd64/). Closing here, thanks for the quick response on this one!

closed 2025-02-03 by Crola1702