ros2/rclcpp #2732: Crash on node shutdown [Jazzy, v28.1.5]

roncapat, 2025-01-31:
There may be a bug on node shutdown, when:
```
use_intra_process_comms: true
start_parameter_services: true
start_parameter_event_publisher: false
```

Here below the stacktrace:

```
#1  0x0000aaaaaf844e10 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0xfffee806c8c0) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#2  0x0000ffff83aa17b4 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0xfffee80fcea0, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:705
#3  std::__shared_ptr<rclcpp::experimental::SubscriptionIntraProcessBase, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0xfffee80fce98, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#4  std::shared_ptr<rclcpp::experimental::SubscriptionIntraProcessBase>::~shared_ptr (this=0xfffee80fce98, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr.h:122
#5  rclcpp::SubscriptionBase::~SubscriptionBase (this=0xfffee80fcd80, __in_chrg=<optimized out>) at /home/user/ros2_jazzy/src/ros2/rclcpp/rclcpp/src/rclcpp/subscription_base.cpp:109
#6  0x0000aaaaaf844e10 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0xfffee80fcd70) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#7  0x0000ffff83ad5cd0 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0xfffee80fc738, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:705
#8  std::__shared_ptr<rclcpp::Subscription<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void>, rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, rclcpp::message_memory_strategy::MessageMemoryStrategy<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0xfffee80fc730, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#9  std::shared_ptr<rclcpp::Subscription<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void>, rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, rclcpp::message_memory_strategy::MessageMemoryStrategy<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > > >::~shared_ptr (this=0xfffee80fc730, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr.h:122
#10 rclcpp::TimeSource::NodeState::~NodeState (this=0xfffee80fc4e0, __in_chrg=<optimized out>) at /home/user/ros2_jazzy/src/ros2/rclcpp/rclcpp/src/rclcpp/time_source.cpp:209
#11 0x0000aaaaaf844e10 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0xfffee80fc4d0) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#12 0x0000ffff83a301c8 in rclcpp::node_interfaces::NodeTimeSource::~NodeTimeSource (this=0xfffee806b7d0, __in_chrg=<optimized out>)
    at /home/user/ros2_jazzy/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_time_source.cpp:53
#13 0x0000ffff83a305e8 in rclcpp::node_interfaces::NodeTimeSource::~NodeTimeSource (this=0xfffee806b7d0, __in_chrg=<optimized out>)
    at /home/user/ros2_jazzy/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_time_source.cpp:53
#14 0x0000ffff839f5b04 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0xfffee810ecd0) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#15 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:705
#16 std::__shared_ptr<rclcpp::node_interfaces::NodeTimeSourceInterface, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#17 std::__shared_ptr<rclcpp::node_interfaces::NodeTimeSourceInterface, (__gnu_cxx::_Lock_policy)2>::reset (this=0xfffee8000c18) at /usr/include/c++/11/bits/shared_ptr_base.h:1272
#18 rclcpp::Node::~Node (this=0xfffee8000b80, __in_chrg=<optimized out>) at /home/user/ros2_jazzy/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:301
```

The affected piece of `time_source.cpp` seems related to the TODO cited in #2370. Do you have any suggestion/hint on how to solve/fix?

fujitatomoya, 2025-02-11:
@roncapat can you provide the self-contained reproducible example for this?

roncapat, 2025-03-10:
Solved, it was not related to rclcpp (nasty linking bug).
Sorry guys for the noise!