ros2/rclcpp #2734: Sub-node name is ignored in service name resolving

PfeifferMicha, 2025-02-06:
When I create sub-nodes using Node::create_sub_node() and then resolve a private name on this sub-node, the sub-namespace is ignored.

```cpp
    auto nh_left = this->create_sub_node( "left" );
    std::string resolved_service_name = nh_left->get_node_base_interface()->resolve_topic_or_service_name( 
                                                                                                                                                   "~/set_camera_info", true, false );
 ```

Resolves to: `/test_cam_info_manager/set_camera_info`
But I would expect it to resolve to: `/test_cam_info_manager/**left**/set_camera_info`

Background:
In order to create two `CameraInfoManager`'s for a stereo setup, I need some way to push the CameraInfoManagers into a sub-namespace. I thought this is exactly what sub-nodes are for, but it's not working. Original issue: https://github.com/ros-perception/image_common/issues/341

Note: In rolling release, I believe this can be circumvented because CameraInfoManager has a new namespace parameter, but this is not the case in jazzy.

**Required Info:**

    Operating System:
        Ubuntu 24.04
    Installation type:
        binaries
    Version or commit hash:
        jazzy
    Client library (if applicable):
        rclcpp

fujitatomoya, 2025-02-06:
@PfeifferMicha thanks for creating issue, this looks like bug.

one question, can you reproduce this issue with rolling build as well? self-contained example would be appreciated.

PfeifferMicha, 2025-02-07:
Thanks for the reply!

Here's my minimal example. Please see the Readme for description of the error/expected outcome.
https://github.com/PfeifferMicha/test_cam_info_manager

I tried with the CI nightly rolling build from here:
https://ci.ros2.org/view/packaging/job/packaging_linux/lastBuild/
Revision: 022da8a9eaa83b4ae8f7b770bdd2a4c700bf1d70

The error is still there.

wjwwood, 2025-02-28:
The reason for this is that the "sub-node" feature exists only in the `rclcpp::Node` class. It does not extend down into the `NodeInterface` classes, therefore the way that the `CameraInfoManager` is using a `NodeInterface` rather than a `Node`. So I don't consider it to be a bug exactly, but just a limitation of the sub-node feature.

You lose the feature when you do `nh_left->get_node_base_interface()`, and it's the same for both Services and Topics.

There's a few options to move forward here:

- you could have the `CameraInfoManager` take a `Node`, which I don't really recommend, because we've been pushing people to take the interfaces instead
- you could extend the sub-node feature down into the node interfaces classes, but that's tricky because you may have multiple sub nodes going at one time, but it could be done I think
- you could extract the sub namespace with `Node::get_sub_namespace()` or similar, and pass it to the `CameraInfoManager` along side the node interface

see: https://github.com/ros2/rclcpp/blob/48a4761faa2ac6339aea45ccd00cd23fd5456d4d/rclcpp/include/rclcpp/node.hpp#L1496-L1554

alsora, 2025-03-13:
Hi @PfeifferMicha , any update on this?

PfeifferMicha, 2025-03-18:
Thanks a lot for the suggestions @wjwwood !
@alsora I've not worked too much with the CameraInfoManager code before, so I have a hard time judging what's the best approach here. I asked people to chime in on the original issue (https://github.com/ros-perception/image_common/issues/341), will wait and see if there's a response.