ros2/rclcpp #2741: Cannot create subscription/publisher with QoS override parameter via parameter callback.

fujitatomoya, 2025-02-12:
### Description

it is not uncommon to create or destroy either subscription or publisher when the parameter is updated.
for example, having `/enable_data_generator` parameter in sensor node, and sensor node watches parameter activity with post set parameter callback using `add_post_set_parameters_callback`. and once that is enabled, sensor node creates the publisher to start publishing sensing data. (or vice-verse)

there is no problem creating publisher or subscription via parameter callback without QoS override parameters for publisher or subscription. but with QoS override parameters, this does not allow us to create publisher or subscription. because parameter set or declare operation is protected by `ParameterMutationRecursionGuard`.

https://github.com/ros2/rclcpp/blob/605251bd71da0427ab2cdca26a9e737ca93e73d9/rclcpp/include/rclcpp/node_interfaces/node_parameters.hpp#L236

this is blocking user application to create either publisher or subscription via parameter callback.

### Reproducible Sample

https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_rclcpp/src/create_sub_via_param.cpp

```bash
### this nodes waits until the parameter update, and then tries to create the subscription during parameter post set callback
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclcpp create_sub_via_param
terminate called after throwing an instance of 'rclcpp::exceptions::ParameterModifiedInCallbackException'
  what():  cannot set or declare a parameter, or change the callback from within set callback
[ros2run]: Aborted

### this command never comes back, since it actually kills the server node
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param set /test_node create_sub true
```

### Additional Information

related to https://github.com/ros2/rclcpp/pull/2378/files#r1411471264

this issue is blocking the enhancement PR https://github.com/ros2/rclcpp/pull/2378,
that using `NodeParameterInterface` instead of `/parameter_event` provides significant efficiency especially ROS 2 system has many nodes in the network. with this fix, we do not need to receive all the events and filter the node and `use_sim_time` parameter from `/parameter_event` topic.

SteveMacenski, 2025-03-07:
Interestingly enough, we just hit this today in Nav2 with a community PR https://github.com/ros-navigation/navigation2/pull/4968

fujitatomoya, 2025-03-07:
@SteveMacenski do you think https://github.com/ros2/rclcpp/pull/2742 makes sense for your use case?

SteveMacenski, 2025-03-07:
I believe so!

closed 2025-04-09 by ahcorde