ros2/rclcpp #2744: rclcpp_components_register_node macro: --no-undefined linker option

ottojo, 2025-02-14:
Hi! When converting nodes to components i found myself running into linker issues of undefined symbols at runtime, which would otherwise occur at compile/link time.

I suggest adding
```cmake
target_link_options(${target} PRIVATE "-Wl,--no-undefined")
```
to the `rclcpp_components_register_node` macro, to ensure there are no undefined symbols when the generated executable is run or the component is loaded into a container later.

I'm unsure of the consequences, or if there are valid use cases for having undefined references in the component library, which is why i want to open this for discussion here first before making a PR. Please let me know if you know of any.
I think users of `rclcpp_components_register_node` would expect to be able to use the generated executable, which would require this anyways.

mjcarroll, 2025-02-14:
It does seem like this would help a class of runtime errors that would be compile/link time errors if you were, for instance, building an executable.

> if there are valid use cases for having undefined references in the component library

I think there are cases, but they would be the rare exception and not the rule.  You can use weak symbols (https://en.wikipedia.org/wiki/Weak_symbol), but this isn't a commonly-used thing that I'm aware of.

Since it could have potential impact, I think that we could tick-tock it:

* Add an-opt in mechanism to the `rclcpp_components_register_node` and leave it off by default.
* Turn the opt-in mechanism on in all of the core places (just to see what happens)
* If there aren't any crazy user-facing impacts, turn it on by default at a later date.

christophebedard, 2025-03-10:
Fixed by #2746.

PR #2746 (2025-03-07): add NO_UNDEFINED_SYMBOLS to rclcpp_components_register_node cmake macro
this adds an option `NO_UNDEFINED_SYMBOLS` to the `rclcpp_components_register_node` cmake macro, which adds the required linker option to the (component library) target to deny undefined symbols.

Undefined symbols usually produce an error if an executable is built, but due to the way the autogenerated executable generated by `rclcpp_components_register_node` is constructed, that doesn't happen here, and the error occurs only when executing it.
This change makes this a link-time error again.

This is especially intended fo [... cut]
PR #2764 (2025-03-11): add NO_UNDEFINED_SYMBOLS to rclcpp_components_register_node cmake macro (backport #2746)
this adds an option `NO_UNDEFINED_SYMBOLS` to the `rclcpp_components_register_node` cmake macro, which adds the required linker option to the (component library) target to deny undefined symbols.

Undefined symbols usually produce an error if an executable is built, but due to the way the autogenerated executable generated by `rclcpp_components_register_node` is constructed, that doesn't happen here, and the error occurs only when executing it.
This change makes this a link-time error again.

This is espec [... cut]
closed 2025-03-10 by christophebedard