ros2/rclcpp #2755: deadline qos for action feedback

asherikov, 2025-02-20:
It feels like deadline QoS may be an interesting option for action feedback, but there seem to be no api to specify QoS event callbacks.

audrow, 2025-02-27:
It'd be great if someone adds this API. If you do, we can review and run tests.

fujitatomoya, 2025-02-27:
just a few comments i have in my mind.

- it cannot always expect the feedback topic is published by action server unless action server is executing the toal. that means once we set the deadline for action feedback topic, and if goal is not executing, there will be QoS deadline event constantly. this could bother the application.
- the application can ignore those events, if the application is not interested in those events. this means, application needs to have some state for that.
- even when the feedback topic is publishing when action server is executing the goal, the question is whose goal is actually executing. feedback topic may be unrelated to this client, but the other. i am not even sure if this is expected behavior.

for me, it would be more useful if we can,
- have distinguished topic instance for each client and server mapping. so that QoS can be set for each mapping instead of affecting other clients.
- have adaptive QoS reconfiguration when the client sends the goal and goal has been accepted on the server. so that we can rely on QoS event that always indicates it misses the expected deadline without any special state control in the application.

Quchaosheng, 2026-10-26:
@audrow @fujitatomoya — picking this up. Before writing an API I want to
answer the three objections from 2025-02-27, because I think they are the
reason this has stayed open, and I think all three are answerable without
changing what `rclcpp_action` is.

## The three objections

**(a) With no goal executing, a deadline is permanently violated.**

The deadline event fires per subscription, not per goal. So yes — a
`deadline` set on the feedback subscription declares "feedback must arrive
constantly", which is false whenever the server is idle. That is not a
detail to document around; it is the wrong event for the use case.

The fix is not to filter it in user code — that is exactly the state
machine @fujitatomoya said should not be pushed onto applications. It is
to expose the event *per goal handle*, and only arm the deadline while that
goal is `ACCEPTED`/`EXECUTING`. rclcpp_action already knows that
transition: `GoalHandle` sets state in `set_state()` on every
`GoalStatus` update, and the client already receives those on the status
subscription. So the arming window is available without new plumbing.

**(b) The feedback topic is shared between clients and between goals.**

`enable_feedback_msg_optimization` already solved half of this: it
configures a content filter on the feedback subscription with the goal ID,
so a client only receives feedback for its own goals
(`client.hpp:204-213`, `client_base.cpp:836-856`). Content filters are not
universally supported, but where they are, "feedback for this client" is
already a real thing.

What remains is "several goals of mine are executing at once, and the
deadline is about goal X". That is why (a)'s answer — deadline scoped to a
goal handle — has to come with the filter, not instead of it.

**(c) Per-client QoS instead of a shared topic.**

This one I would not try to solve. Separate topic instances per
client↔server pair would change the action protocol's discovery and
remapping surface, which is not a QoS change. The content filter is the
mechanism that already exists for this and it does not need a new topic
name.

## The blocker I hit that is worth deciding first

There is no place to pass feedback QoS today. The action client's
subscriptions are created by `rcl_action_client_init()` from a single
`rcl_action_client_options_t`, and `ClientBase`'s constructor takes that
struct directly:

```
include/rclcpp_action/client.hpp:145
  const rcl_action_client_options_t & client_options = rcl_action_client_get_default_options(),
```

`rcl_action_client_options_t` carries one `qos` for all five endpoints
(goal service, result service, cancel service, feedback topic, status
topic). `rclcpp_action` only overrides the *status* profile, via
`DefaultActionStatusQoS` (`rclcpp_action/src/qos.cpp`). So:

- setting `deadline` through the existing struct also applies it to the
  goal/result/cancel services and the status topic, which is not what
  anyone wants;
- `QoS` event callbacks (§4.5) are set on the *subscription*, and the
  feedback subscription is built inside rcl, not by rclcpp. Even with the
  right depth, there is currently no rclcpp-side object to call
  `set_event_callbacks()` on.

So this needs a small `rcl` addition — a way to set per-endpoint QoS
options before `rcl_action_client_init()` — or `rclcpp_action` has to own
the feedback subscription. The second is a much bigger change. I would
rather see the first, and it would unblock per-endpoint QoS generally,
not just deadline.

## Proposed shape

```cpp
// rclcpp_action/qos.hpp
class FeedbackQoS : public rclcpp::QoS { /* deadline/liveliness are opt-in */ };

// client.hpp, ClientOptions
struct ClientOptions
{
  // ... existing fields ...
  // Feedback QoS; services and the status topic keep their defaults.
  // An unset deadline preserves today's behaviour.
  std::optional<rclcpp::QoS> feedback_qos = std::nullopt;
  // Called when feedback for THIS client's armed goal misses its deadline.
  // Empty by default; never called while no goal of this client is executing.
  std::function<void(const GoalHandle::SharedPtr &)> feedback_deadline_callback = nullptr;
};
```

Semantics I would write down before implementing:

1. The callback is only armed while a goal of this client is in
   `ACCEPTED`/`EXECUTING`, and it names the goal it fired for. Idle
   clients never see it.
2. If several goals are executing, the callback may fire for any of them;
   it is not a per-topic event anymore, so this has to be explicit.
3. Nothing fires between `EXECUTING` and the result arriving — the server
   is allowed to be quiet while finalizing, and a deadline there would be
   noise.
4. Losing the deadline is not an error state: it reports, it does not
   cancel the goal. `rcl`'s deadline event is a notification, and making
   the action layer turn it into cancellation would be a much larger
   semantic change.

## What I am not proposing

Not deriving the deadline from `sensored`/`KeepLast` depth, and not
letting the executor or DDS decide. The user sets the expected feedback
period because only the user knows the controller's rate.

If the `rcl` per-endpoint QoS change is acceptable in principle, I will
write the `rcl` piece first and the rclcpp API on top of it — that way the
rclcpp part is testable with a plain subscription before any action
plumbing lands. If there is a reason to keep `rcl_action_client_options_t`
as-is, say so now and I will look at owning the feedback subscription in
rclcpp instead, though I expect that to be rejected for good reasons.