ros2/rclcpp #2756: Topic Statistics for multiple subscriptions in same node aren't differentiated

emersonknapp, 2025-02-21:
When you enable topicstatistics, the `/statistics` messages that are published only contain the node's name, not the subscription/topic - so when you turn on this feature for a node with multiple subscriptions, the statistics coming out can't be interpreted meaningfully.

kphan20, 2025-03-09:
I would like to help with this issue. I haven't contributed before but would like to try. Is this a good issue to start with?

fujitatomoya, 2025-03-10:
this is so true. without having the topic information in statistics message, we cannot tell whose statistics at all... 😓

christophebedard, 2025-03-12:
> I would like to help with this issue. I haven't contributed before but would like to try. Is this a good issue to start with?

@kphan20 happy to hear that you want to contribute! I think this issue is a bit tricky, since we first need to decide on a solution. I would recommend looking for another/more straightforward issue to get started. You can search for issues with the "good first issue" label: https://github.com/search?q=org%3Aros2+label%3A%22good+first+issue%22+++&type=issues&state=open