ros2/rclcpp #2757: rclcpp_action std::runtime_error when killing action server

fafux, 2025-02-26:
Hi, I am working on Ubuntu 24.04 with ROS2 jazzy.
I followed the tutorials on how to ["Writing an action server and client (C++)"](https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Writing-an-Action-Server-Client/Cpp.html). I launched the server and then the client as written in the tutorial. If I send a SIGINT to the server while the action is executing (^C) I get an std::runtime_error specifically: 
```
[INFO] [1740584812.433644455] [rclcpp]: signal_handler(signum=2)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Asked to publish result for goal that does not exist
```
I think this is a bug. The runtime error is thrown before rclcpp::shutdown() is called. As far as I know the server should send a notification to the client that the goal was canceled or aborted, call rclcpp::shutdown(), and finish cleanly. The problem is that, with this bug, the resources are not released.

sloretz, 2025-03-06:
Just adding the location of the exception: https://github.com/ros2/rclcpp/blob/48a4761faa2ac6339aea45ccd00cd23fd5456d4d/rclcpp_action/src/server.cpp#L733

fujitatomoya, 2025-03-07:
about the exception on action server, i think you met the racy condition. I cannot reproduce this issue though.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run action_tutorials_cpp fibonacci_action_server
[INFO] [1741309035.485585161] [fibonacci_action_server]: Received goal request with order 10
[INFO] [1741309035.486094219] [fibonacci_action_server]: Executing goal
[INFO] [1741309035.486319267] [fibonacci_action_server]: Publish feedback
[INFO] [1741309036.486456961] [fibonacci_action_server]: Publish feedback
^C[INFO] [1741309037.169525708] [rclcpp]: signal_handler(signum=2)
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run action_tutorials_cpp fibonacci_action_server
[INFO] [1741309042.508744397] [fibonacci_action_server]: Received goal request with order 10
[INFO] [1741309042.509333889] [fibonacci_action_server]: Executing goal
[INFO] [1741309042.509530099] [fibonacci_action_server]: Publish feedback
[INFO] [1741309043.509735177] [fibonacci_action_server]: Publish feedback
^C[INFO] [1741309044.408719362] [rclcpp]: signal_handler(signum=2)
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run action_tutorials_cpp fibonacci_action_server
[INFO] [1741309049.774068952] [fibonacci_action_server]: Received goal request with order 10
[INFO] [1741309049.776226357] [fibonacci_action_server]: Executing goal
[INFO] [1741309049.777100021] [fibonacci_action_server]: Publish feedback
[INFO] [1741309050.776877790] [fibonacci_action_server]: Publish feedback
^C[INFO] [1741309051.480142554] [rclcpp]: signal_handler(signum=2)
```

> The runtime error is thrown before rclcpp::shutdown() is called.

i think `Context::shutdown()` is called by deferred signal handler when the SIGINT comes in, it shuts down all the context gracefully.

> As far as I know the server should send a notification to the client that the goal was canceled or aborted

this seems like a bug, unexpected behavior.
action client is waiting for the result, and action server actually tries to publish `action_msgs::msg::GoalStatus::STATUS_CANCELED` during destruction.

https://github.com/ros2/rclcpp/blob/48a4761faa2ac6339aea45ccd00cd23fd5456d4d/rclcpp_action/include/rclcpp_action/server_goal_handle.hpp#L246-L251

closed 2025-09-16 by fujitatomoya