ros2/rclcpp #2760: rclcpp failed to send response to service client

URJala, 2025-03-04:
When calling a service I will sometimes (quite often), get an error message:
`[controller_manager.rclcpp]: failed to send response to /controller_manager/list_controllers (timeout): client will not receive response, at ./src/rmw_response.cpp:153, at ./src/rcl/service.c:400`
This happens sometimes with any service I call, but especially `list_controllers` and `switch_controllers` seem to throw this error a lot (but not every time). When it happens the service will just hang, presumably because it never received a response.
It is specifically happening while i am using the Universal Robots ROS2 driver, but I don't think that is related.
I am running the driver (and therefore ROS2) in a vscode devcontainer (docker container), and i dont know if that could be related?
I am running ROS Jazzy, built from binaries (the devcontainer is built with `FROM ros:jazzy)`.

fujitatomoya, 2025-03-04:
the error shows that underlying rmw implementation (i am assuming you are using rmw_fastrtps) cannot send the request response to the service client with timeout, because service client may not exist anymore. can you make sure service client still exists until it receives the request response? or if this is initialization phase, it would be nice to check if the service server is online before sending any requests to the server. this makes sure service topics are available before starting the communication.

https://github.com/ros2/rclcpp/blob/48a4761faa2ac6339aea45ccd00cd23fd5456d4d/rclcpp/include/rclcpp/client.hpp#L206-L219

and self-contained reproducible example would be useful for further debug.

zacharyyamaoka, 2025-04-24:
I am having the same issue!

In this setup I have 1 rclpy node that is interacting with two rclcpp server nodes via a service interface 

I call the service using `_call_async`. I sometimes run the rclpy node as a `multithreadedexecutor()` or sometimes I do a scripting setup where there is no executor and I manually spin with `spin_until_future_complete`.

I experince the problem in both cases.

```python
    def _call(self, request, timeout_sec=5):

        future = self._call_async(request)

        if self.spin:
            # Spin in this thread until the service call is complete
            rclpy.spin_until_future_complete(self.node, future, timeout_sec=timeout_sec)
        else:
            # Manually wait using rate, assuming another thread is spinning already
            start_time = time.time()
            while not future.done():
                if time.time() - start_time > timeout_sec:
                    self.log.warn("Service call timed out.")
                    future.cancel()
                    return None
                self.wait_rate.sleep()

        if future.result() is None:
            if self.spin:
                self.log.warn(f"spin_until_future_complete() timed out")

            elif future.exception() is not None:
                self.log.warn(f"Service call returned but failed with exception: {future.exception()}")
            else:
                self.log.warn("Service call returned but with result.")
            return None

        response = future.result()
        self._process_response(request, response)
        return response

    def _call_async(self, request):

        self.log.debug("Making service call")
        self.send_time = time.time()

        if self.bam_api:
            request.header.client_id = self.client_id
            request.header.stamp = self.node.get_clock().now().to_msg()

        self.future = self.service_client.call_async(request)
        return self.future

    def _process_response(self, request, response):
        self.last_request_duration = time.time() - self.send_time
        self.log.info(f"Service call complete in {self.last_request_duration:.3f}s")
```

 Every once in a while, the server will fail to return with error msg `failed to send response to /bam_GPU/get_joint_state (timeout): client will not receive response, at ./src/rmw_response.cpp:153, at ./src/rcl/service.c:400`

```
]: Returning Current Joint State
[joint_state_buffer-4] [INFO] [1745506222.797931390] [bam_GPU.joint_state_buffer_node]: Returning Current Joint State
[joint_state_buffer-4] [INFO] [1745506227.050707006] [bam_GPU.joint_state_buffer_node]: Returning Current Joint State
[joint_state_buffer-4] [INFO] [1745506347.828642937] [bam_GPU.joint_state_buffer_node]: Returning Current Joint State
[joint_state_buffer-4] [INFO] [1745506350.719951926] [bam_GPU.joint_state_buffer_node]: Returning Current Joint State
[joint_state_buffer-4] [WARN] [1745506350.820081571] [bam_GPU.joint_state_buffer_node.rclcpp]: failed to send response to /bam_GPU/get_joint_state (timeout): client will not receive response, at ./src/rmw_response.cpp:153, at ./src/rcl/service.c:400
[joint_state_buffer-4] [INFO] [1745506651.977681875] [bam_GPU.joint_state_buffer_node]: Returning Current Joint State
[joint_state_buffer-4] [INFO] [1745506669.794551165] [bam_GPU.joint_state_buffer_node]: Returning Current Joint State
```

[... 171 lines cut ...]
```

Often when if I start up the rclpy node and just make one request, the calls seem to take longer/hang a bit. I think this is to do with performance issues of the python multithreaded executor...

```
[INFO] [1745507126.295464171] [bam_GPU.arm_server.move_it_client]: Sending planning request
[INFO] [1745507126.296336479] [bam_GPU.arm_server.joint_buffer_client]: Service call complete in 0.000s
[INFO] [1745507126.924803121] [bam_GPU.arm_server.kinematic_path_client]: Service call complete in 0.628s
[INFO] [1745507126.925201312] [bam_GPU.arm_server.move_it_client]: Planning request complete
```

To sumarise, the server will throw the error:
```
[move_group-2] [WARN] [1745507213.868207181] [bam_GPU.move_group.rclcpp]: failed to send response to /bam_GPU/plan_kinematic_path (timeout): client will not receive response, at ./src/rmw_response.cpp:153, at ./src/rcl/service.c:400
```

but the client is definetly still running! Actually it ends up timing out beacuse the server didn't response....

```
[INFO] [1745507213.128556331] [bam_GPU.arm_server.move_it_client]: Sending planning request
[INFO] [1745507213.129553407] [bam_GPU.arm_server.joint_buffer_client]: Service call complete in 0.001s
[WARN] [1745507218.130359959] [bam_GPU.arm_server.kinematic_path_client]: spin_until_future_complete() timed out
[WARN] [1745507218.130719206] [bam_GPU.arm_server.kinematic_path_client]: Planning failed! and no result recivied
[INFO] [1745507218.131002929] [bam_GPU.arm_server.move_it_client]: Planning request complete
```

zacharyyamaoka, 2025-04-24:
Reflecting on this further.

I think the issue is caused by not giving the lower level systems enough time to start up.

So far, **the issue either happens on the first service call, or never again.**

I haven't yet experienced it working, and then stop working, then continue working

I think using a small sleep before making the first service call can be the solution....

I will update with my findings as I continue.

jmachowinski, 2025-09-01:
Uhm, this looks like the wait_for_service call is missing before doing the first call...

DaniDoro, 2025-09-01:
This is still an issue despite using wait_for_service. I've managed to recreate it several times with ros2 jazzy and cyclone dds.

jmachowinski, 2025-09-02:
Might it be that old processes from the previous launch are still lurking around ?

DaniDoro, 2025-09-02:
That has also been considered. However, the re-launching is done as a systemd service that is restarted, and that ensures that the former process is terminated.