ros2/rclcpp #2763: Publisher cleanup via service call results in deadlock on Jazzy

Serafadam, 2025-03-10:
Relevant issue here:
https://github.com/luxonis/depthai-ros/issues/657
Short description:
When using a service for resetting the driver's state (mostly cleaning up classes that have publishers inside them), the driver has a high chance to hang without response in rclcpp::Executor::wait_for_work (rcl_wait/rmw_wait). This was thought to be fixed by assigning Reentrant CB groups to services that correspond that call this cleanup, but it seems that it didn't help in the end.
The issue has been observed on both Cyclone and FastDDS RMW.
Humble and Noetic versions of the driver don't experience this issue.

fujitatomoya, 2025-03-10:
@Serafadam can you try with jazzy source build https://github.com/ros2/rclcpp/tree/jazzy? we have fixed several bugs related to executors, so i think it would be worth to give it a shot if you have not done that.
if you still see the problem with rclcpp jazzy source build, could you provide the minimal self-contained reproducible example?

Serafadam, 2025-03-11:
Hi, thanks for the fast response, the issue seems to be solved when using latest commit, I'll also try to verify with the customer reporting the issue

fujitatomoya, 2025-03-11:
@Serafadam that is great to hear, and will be waiting for the update! thanks for taking your time to verify 👍