ros2/rclcpp #2777: Improved docstring for `get_rmw_qos_profile()`

abhishek47kashyap, 2025-03-25:
The current docstrings for `get_rmw_qos_profile()` don't provide any more information than the obvious. How to make them better? I know next to nothing about middlewares, but I'm happy to submit improvements if they are welcome.
https://github.com/ros2/rclcpp/blob/1564fc23c62d7966db835c3e5a4c6e3ad0012288/rclcpp/include/rclcpp/qos.hpp#L148-L154

---

**(my) Motivation**: I was going through the [tutorial](https://docs.ros.org/en/humble/p/message_filters/doc/Tutorials/Approximate-Synchronizer-Cpp.html) on Approximate Time Synchronizer and noticed these lines:
```
  rclcpp::QoS qos = rclcpp::QoS(10);
  temp_pub = this->create_publisher<sensor_msgs::msg::Temperature>("temp", qos);
  fluid_pub = this->create_publisher<sensor_msgs::msg::FluidPressure>("fluid", qos);

  temp_sub.subscribe(this, "temp", qos.get_rmw_qos_profile());
  fluid_sub.subscribe(this, "fluid", qos.get_rmw_qos_profile());
```
I wanted to find out more on `get_rmw_qos_profile` and found the docstring not telling me much. The [documentation](https://docs.ros.org/en/rolling/p/rclcpp/generated/classrclcpp_1_1QoS.html) is not more informative either.

mjcarroll, 2025-03-25:
The `get_rmw_qos_profile` is intended to return a handle to the underlying `rmw_qos_profile_t` structure from the middleware.  This structure is defined in `rmw` here: https://github.com/ros2/rmw/blob/713d0ad3779413764e929141cf49bb9aef93b82c/rmw/include/rmw/types.h#L569-L614

There is a set of "standard" QoS profiles that could be use depending on the application area: https://github.com/ros2/rmw/blob/713d0ad3779413764e929141cf49bb9aef93b82c/rmw/include/rmw/qos_profiles.h  Each of those profiles is "wrapped" in a C++ class in the bottom of the QoS header that you linked.

I think in this case, the documentation actually predates the ability to pass an `rclcpp::QoS` in to the `subscribe` method in the `message_filters::Subscriber`. If you use this signature: https://github.com/ros2/message_filters/blob/65746bda70ff93ee7a981997c2783764a67045c3/include/message_filters/subscriber.hpp#L61-L63

This could be rewritten to simply be:

```  
rclcpp::QoS qos = rclcpp::QoS(10);
temp_pub = this->create_publisher<sensor_msgs::msg::Temperature>("temp", qos);
fluid_pub = this->create_publisher<sensor_msgs::msg::FluidPressure>("fluid", qos);

temp_sub.subscribe(this, "temp", qos);
fluid_sub.subscribe(this, "fluid", qos);
```

Would you be interested in verifying this and submitting an update to the tutorial?

fujitatomoya, 2025-03-26:
actually this signature is deprecated, and will be removed in after next release.

https://github.com/ros2/message_filters/blob/65746bda70ff93ee7a981997c2783764a67045c3/include/message_filters/subscriber.hpp#L114-L126

tutorial needs to be update with supported signature as mentioned above.

@abhishek47kashyap i think your point is to add more docstring for `get_rmw_qos_profile`, please feel free to open the PR to improve that.

fujitatomoya, 2025-03-27:
> tutorial needs to be update with supported signature as mentioned above.

https://github.com/ros2/message_filters/pull/127 has addressed documentation as well. (not humble because signatures with rmw are deprecated in rolling only.)

abhishek47kashyap, 2025-03-29:
@mjcarroll @fujitatomoya PR just went up, would be great to get it reviewed 🙂

closed 2025-03-30 by fujitatomoya