ros2/rclcpp #2793: create_timer throws but isn't documented to throw

Ryanf55, 2025-04-01:
### Operating System:

Ubuntu 22

### ROS version or commit hash:

Jazzy

### RMW implementation (if applicable):

fastrtps

### RMW Configuration (if applicable):

N/A

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

```Formatted
N/A - it's proprietary
```

### Steps to reproduce issue

1. Create a node, and add a (while/for) loop call that blocks while rclcpp::ok() is true
2. Trigger a ctrl+C on the node which now means that rclcpp::ok() is false
3. Later in code execution, call create_timer(..
4. Observe the timer creation fail and throw
5. Observe a non-zero exit code from the node

### Expected behavior

If exceptions are throw, they are documented in the header

### Actual behavior

This exception is not documented

### Additional information

```
[Thread 0x7fff5e7fc6c0 (LWP 1030613) exited]
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  Couldn't initialize rcl timer handle: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/guard_condition.c:67

Thread 1 "MyNode" received signal SIGABRT, Aborted.
```

Here's the backtrace.

[image]

If you need a minimum reproducible example, I can help provide one, but I haven't spent the time to write one yet.

jmachowinski, 2025-04-02:
I feel like this is a user error. You should register a shutdown handler and terminate your application. It's kind of obvious to me that the further usage of the system throws exceptions around if you try to use it afterwards. Actually the message in the exception is exactly stating whats going on...

I don't follow the it should be documented argument here. If I follow this logic, I would also need to add a documentation on every method, that it could throw std::bad_alloc if the system runs out of memory.

fujitatomoya, 2025-04-02:
@Ryanf55 i am actually with @jmachowinski on this, i do not think this is bug.

1st of all, this is expected and designed exception. i think user application needs to take care of this exception.

IMO, the docstring for a method or function should describe the exceptions that this method itself explicitly raises. It is not necessary to document exceptions that might be raised by functions or methods called within this method, unless the exception is propagated directly in the method. This approach keeps the documentation focused and avoids overwhelming the user with unnecessary details.

Ryanf55, 2025-04-02:
Ok fair enough. Sounds like people generally ignore coding for a clean shutdown. Unless we wrap every call to rclcpp, such as `publish` with a check to rclcpp::ok or catching an exception, then a throwing application that uses SHM as transport can corrupt the file handlers and require a hardware reboot. That's why we are trying to clean this up. We have a ROS node that we stop at runtime on a system, and if it can't exit cleanly, it breaks DDS comms.

Is there an example of production open source ROS code that checks rclcpp::ok or catches exceptions and cleanly shuts down on every interraction with the middleware?

christophebedard, 2025-04-02:
> I don't follow the it should be documented argument here. If I follow this logic, I would also need to add a documentation on every method, that it could throw std::bad_alloc if the system runs out of memory.

I agree with you, but it might be worth documenting it _somewhere_, since this is the case for pretty much all objects. Not sure where, though.

jmachowinski, 2025-04-04:
> Is there an example of production open source ROS code that checks rclcpp::ok or catches exceptions and cleanly shuts down on every interraction with the middleware?

If you use a single threaded exectutor, you should be fine, if you shutdown as last action in any entity callback. After this the executor will terminate automatically.

If the shutdown comes in asynchronously by some sort of callback, stop the executor and wait for it to finish and shutdown afterwards.

For multithreaded executors, this comes down to, clean up all your entities that are known to run in a parallel callback group then shut down.