ros2/rclcpp #2794: Double spin still required in 29.3.0

ajtudela, 2025-04-04:
### Operating System:

Ubuntu 24.04

### ROS version or commit hash:

rolling from binaries (29.3.0)

### RMW implementation (if applicable):

rmw_cyclonedds_cpp

### Client library (if applicable):

rclcpp

### 'ros2 doctor --report' output

### Steps to reproduce issue

callback_group_executor_.spin_some()

### Expected behavior

Spin_some work as previously expected.

### Actual behavior

Spinning some requires a warmup spin that doesn't do anything before processing useful work. A double spin is required to get the work done.

### Additional information

While testing Nav2 (`main` branch) last morning, I noticed that a BT node it wasn't working as expected. After checking the code, I noticed that the callback was not being executed with this [spin_some](https://github.com/ros-navigation/navigation2/blob/08268f9922f92511f12ad9b951148d1c1d30096b/nav2_behavior_tree/plugins/condition/is_battery_charging_condition.cpp#L67). 

I opened an issue in Nav2 for more information [here](https://github.com/ros-navigation/navigation2/issues/5041).

But it's related to this rlccpp issue #2589. I thought it was fixed in #2724 and #2729, but I still have this problem in rolling (and jazzy).

jmachowinski, 2025-04-04:
29.3.0 is from december, the fix was merged in januar. 

Please reopen the issue, if you can still reproduce it with the latest source version.

@mjcarroll Do we need to also produce a rolling release ?

closed 2025-04-04 by jmachowinski

mjcarroll, 2025-04-04:
> Do we need to also produce a rolling release ?

Looks like the last release was in December, so I can get another one out.

jmachowinski, 2025-04-07:
Did you recompile your code base ? The fix was in template code. You may need to force the build.

jmachowinski, 2025-04-07:
Okay, that's super strange, can you give me a minimal reproducer ?

jmachowinski, 2025-04-07:
Actually looking at the code you are referencing, I would say, you want to use spin_all, not spin_some.